MarlinFirmware/Marlin #22839: [BUG] Part cooling fan not working at output 7

michal777, 2021-09-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I uploaded marlin to my Renkforce RF100 XL, did necessary modifications and then noticed that part cooling fan doesn't work.

### Expected behavior

Part cooling fan should work when #define FAN_PIN  7. Corresponding LED should be continuously on after enabling instead of dim flashes.

### Actual behavior

If I set pins that drive part cooling fan and hot-end cooling fan as follows:
`#define FAN_PIN  7`
`#define E0_AUTO_FAN_PIN  9`
the result is: fan for hot-end OK but part cooling fan not running. Corresponding LED shows that the MOSFET is driven with very tiny voltage of PWM (weakly flashes) when fan enabled.

Test to confirm hardware is OK fans swapped:
`#define FAN_PIN  9`
`#define E0_AUTO_FAN_PIN  7`
the result is: heater cooling and part cooling fans are both working and I can control them using GUI, LED shows all MOSFETS are driven OK.

### Steps to Reproduce

1. I started with ready configuration for my renkforce rf100 xl
2. My modifications of Configurations-bugfix-2.0.x/config/examples/Renkforce/RF100XL/Configuration.h:
-#define MOTHERBOARD BOARD_RAMPS_14_EFB
+#define MOTHERBOARD BOARD_MKS_BASE_15
-#define FAN_PIN  -1
+#define FAN_PIN  7
-#define INVERT_X_DIR false
-#define INVERT_Y_DIR true
+#define INVERT_X_DIR true
+#define INVERT_Y_DIR false
-#define LCD_BED_LEVELING
+//#define LCD_BED_LEVELING
3. Upload firmware and try to enable fan through the printer's GUI. Observe LED on board - it flashes weakly but not enough to provide power to the fan.
4. Optionally swap pin numbers as below and check that both fans can be controlled and work as expected although opposite way
#define FAN_PIN  7
#define E0_AUTO_FAN_PIN  9

### Version of Marlin Firmware

2.0.3 (i'm not sure if controller or LCD module version is displayed in "about" :( )

### Printer model

Renkforce RF100 XL

### Electronics

MKS base v1.5

### Additional information & file uploads

[image]
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7230460/configuration.zip)

ellensp, 2021-09-26:
Your fan_pin and fan1_pin are the same.
This is a conflict

You added #define FAN_PIN  7 to Configuration.h
The motherboard make marlin use Marlin/src/pins/ramps/pins_MKS_BASE_15.h this includes Marlin/src/pins/ramps/pins_MKS_BASE_common.h
which has  #define MOSFET_D_PIN  7 then it includes  Marlin/src/pins/ramps/pins_RAMPS.h
which sets  #define FAN1_PIN  MOSFET_D_PIN

Looking for away around this. (without making code changes)

ellensp, 2021-09-26:
Add #define IS_RAMPS_EFB to Your Configuration.h 
This should fix it.

michal777, 2021-09-26:
It works :D

btw, if I find issues with the configuration file for my printer, should I open a new issue?
the problems are:
- inverted X and Y
- motherboard type (although maybe not an issue?)
- compilation fails with LCD_BED_LEVELING so I disabled it
- the part cooling fan problem which is already solved (#define FAN_PIN 7 and #define IS_RAMPS_EFB)

ellensp, 2021-09-26:
You should open an issue in Configurations. https://github.com/MarlinFirmware/Configurations

ellensp, 2021-09-26:
The issue is often manufacturers change parts, without telling anyone.
I suspect this really did have a ramps in it, when it was release. Most listed issue comes from swapping to a BOARD_MKS_BASE_15

closed 2021-09-26 by ellensp