MarlinFirmware/Marlin #22845: [FR] (prevent fire - disable heaters after x minutes of inactivity)

xjustbmw, 2021-09-27:
### Is your feature request related to a problem? Please describe.

turn heaters off , after x minutes of serial connection inactivity

### Are you looking for hardware support?

ender 3 v1 
raspberry pi 4

### Describe the feature you want

using octoprint session if the usb connection is interrupted, the printer remains with both heaters on , can we add a feature to disable them after x minutes of inactivity ? to prevent any fire in case this happens again

### Additional context

i suspect in my case there is a bad psu on either the raspberrypi4 or the printer's psu.  disconnection happens when i stop another electrical appliance nearby.

ellensp, 2021-09-27:
Its already there. I Configuration_adv.h
```
/**
 * Hotend Idle Timeout
 * Prevent filament in the nozzle from charring and causing a critical jam.
 */
//#define HOTEND_IDLE_TIMEOUT
```

closed 2021-09-27 by ellensp

xjustbmw, 2021-09-27:
oh. how did i miss it during compilation
thanks a lot ellen