MarlinFirmware/Marlin #22849: [FR] (Electrochemical Machining (ECM) and EDM Retraction Support)

JohnAtMorlock, 2021-09-28:
### Is your feature request related to a problem? Please describe.

Me and several others have been using Marlin for our ECM (electrochemical machining) CNC machines. ECM Allows us to cut metal without using any physical force. (https://youtu.be/KieJN-J4s38) 

ECM is sort of like EDM, in that current must be kept constant by carefully controlling how far away our tooling is from the work. Usually, this is a few hundred microns. If the tool gets too close to the work, it shorts. If it is too far away, the current drops.  

We need to run the printer backwards when the current gets too high or drops to zero (shorts)

### Are you looking for hardware support?

Cheap current sensors report their values in the form of an analog value between 0 and 5 volts. All that is required is a way to declare a "safe" range of analog values

Example) 
Minimum_Sensor_Value = 1023;
Maximum__Sensor_Value = 1023;

The value/current relationship changes from sensor to sensor

### Describe the feature you want

We require a feature that detects when a sensor reports the current is too high or low. Similar to a filament runout sensor in theory, the behavior of the printer is controlled by an external stimuli. 

Theory of operation: 
1) A sensor pin receives an analog value from a current sensor 
2) If the analog value is too high or too low, an event is triggered
3) The event is as follows: The machine reverses its last move.
    G1 Z0.05 F600 => G1 Z-0.05 F100
The reversal is continued until the sensor value is within the declared range. 
4) The program resumes at the line of gcode it first started reversing at 

Example)
Move to point A
Move to point B
Move to point C
Move to point D
Move to point E 
DETECT CURRENT IS TOO HIGH 
Move to point D
STILL DETECT CURRENT IS TOO HIGH 
Move to point C 
CURRENT RETURNED TO NORMAL 
Move to point D 
Move to point E
Move to point F

Essentially, the machine needs to run its gcode backwards until the sensor reading goes back to normal. This would be conducted differently depending upon if the machine was in relative or absolute positioning mode. It would also need to search for not only G1 commands, but also E1 since we use the E-axis on our machines as a cheap and easy rotational axis.

DerAndere1, 2021-09-28:
Marlin now has support additional axes, see https://github.com/MarlinFirmware/MarlinDocumentation/pull/409/files or https://github.com/DerAndere1/Marlin/wiki, so you do not have to use extruders for the rotational axes and you can add endstops to them. That approach may be a little more robust long term. Can you point me to code that calculates the required rotational speed for the rotational axes depending on the current position of the other axes position to opbtain constant feedrate?

JohnAtMorlock, 2021-09-28:
We have not been calculating the required rotational speed. I just tell tell the rotational axis to move to a certain position using hand written gcode. It is all touch and go. 

G1 E90 sort of thing

DerM4209, 2021-11-11:
Hi,
I'm currently working on an EDM machine and I would also need this feature for it.
My idea of doing this without the feature would be running the GCODE with fast feedrate on the machine controller and recording the Step / Dir signals of all axis with a second microcontroller. After that when the GCODE has finished I would playback the recorded Step / Dir signals again and apply the voltage reading on it for reversing it when needed.
But it would be really cool if Marlin would support such a feature.
Has anybody an idea how to start developing something like that for Marlin?

DerM4209, 2021-11-30:
While trying to figure out how marlin works, I had an idea how it could be done without saving the position in intervals, what likely would cause some problems.
Maybe it could be done by saving just two positions for G1 commands (a starting position and a target position) and moving back and forth between them until the material got removed. Like:
start print/machining job
receive G1 X Y Z code and send busy to host
save current position as starting position
save position from G1 command as target position
move towards target position until machine detects a short
stop moving and move instead towards starting position
when there is no longer a short stop moving and move instead towards target position again
when the machine reaches target position save target position as starting position and receive a new G1 command
and so on...

DerAndere1, 2021-12-02:
Currently G1 first uses `get_destinatio_from_command()` to update the `destination`variable. Then it calls https://github.com/MarlinFirmware/Marlin/blob/72b99bf1ba24cb9124668b958039b32a164c68cd/Marlin/src/module/motion.cpp#L1239

For ECM, G1 would have to call a modified version of that function where the arc voltage has to be monitored. Voltages below a configurable threshold should limitat the actual feedrate (which probably means a recalculation of the planned trajectory (see planner.cpp)). The polling of the arc voltage could be done as it is currently done for endstops: Endstops are implemented in  [endstops.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/endstops.cpp and https://github.com/MarlinFirmware/Marlin/blob/b9ee84e8dcc6ae81aba3d46e079ed26e1a308a49/Marlin/src/module/planner.cpp#L1708), the endstop polling is called from the temperatur ISR: https://github.com/MarlinFirmware/Marlin/blob/b9ee84e8dcc6ae81aba3d46e079ed26e1a308a49/Marlin/src/module/temperature.cpp#L3518

DerM4209, 2021-12-02:
@DerAndere1, thank you very much for your reply, I will look at the files you mentioned.
For the case that a short occurs, do you think moving back towards the target position of the last G1 command would be a suitable aproach or could there be a better way to do it?

DerAndere1, 2021-12-03:
I have no clue about ECM. But from your link to the source code for the control loop (https://www.dynomotion.com/forum/viewtopic.php?f=7&t=680#p1369) you can see that there are three conditions: 
```
if (Voltage lower than SC voltage with some safety limit) 
  go reverse
if (Voltage higher than Upper GAP window)
  go faster
if (Voltage higher than SC but lower than Down GAP window)
  go slower
```

DerM4209, 2021-12-06:
I could somehow figure out how to get a voltage reading from temperature.cpp and report it to serial. So, in the next step I can maybe try to check and report the voltage range and find a way to modify the G1 command, so that it can react to that.

DerM4209, 2021-12-15:
Hi,
I got some code ready for testing, so that I could drill single holes with it on my EDM setup.
I uploaded the changed files to my github profile, but the code is not finished, yet.

Here is some footage of it:
https://youtu.be/BT2hdXJYGeE

At the moment, it is just for a single move and missing a gcode command to turn it on and off, but I will add that soon.

DerAndere1, 2021-12-16:
Congrats! I was able to find changes in temperature.h and temperature.cpp by searching for "EDM" and "edm".

DerM4209, 2021-12-20:
Thank you :)
While working on the code, I noticed some strange  behavior when using the get_cartesian_from_steppers() function to get the cartes variable. I used cartes to compare it with my target variable to check whether the machine has reached the target.
My problem is:
When I run the code I can drill in millimeter steps, but when I try to drill in sub - millimeter 0.10mm
 steps, start from a position with decimal places or enter multiple commands in fast succession, the cartes variable gets no longer updatet / the machine never reachs the target.

Do you know what could be the cause, or could it even be a bug in the get_cartesian_from_steppers() function?

Is there another way to check whether the machine has really reached the target?

DerAndere1, 2021-12-20:
I did not spend a lot time tinkering about this, and I guess I guess I missed something fundamental. But in my understanding, Marlins loop() will be called after initialization, and loop() calls idle(). idle() calls ecm_edm_mode(), which (as long as the target is not reached) calls move_to_target(), which then calls idle(). So as long as target is not reached, you have an "inner loop" there and I think none of the other tasks of idle() are executed during this time:
 *  - Handle filament runout sensors
 *  - Run HAL idle tasks
 *  - Handle Power-Loss Recovery
 *  - Run StallGuard endstop checks
 *  - Handle SD Card insert / remove
 *  - Handle USB Flash Drive insert / remove
 *  - Announce Host Keepalive state (if any)
...

also, the call to `get_cartesian_from_steppers();` in move_to_target() is outside of this "inner loop", so it will not get called. This may explain that the `cartes` is not updated. To fix it, in file ecm_edm_mode.h, try to swap lines 18 and 19: 
```
idle();
get_cartesian_from_steppers();
``` 
becomes this
```
get_cartesian_from_steppers();
idle();
```

DerAndere1, 2021-12-20:
Also research how to check for equality of floating point numbers on computers. Marlin has a macro for this problem, so
```
if (cartes == target_position) {
```
should be replaced by 
```
if (NEAR(cartes, target_position)) {
```

That might get rid of the second bug with decimal starting positions

DerM4209, 2021-12-20:
Thank you for your help, I tried to change the code like you said, but the problem still exists.
I also reduced the code to a minimum, to see whether it could be caused by something else, but nothing changed.

It is easy to reproduce, moving in XY in 10mm or 1mm steps is not a problem, but as soon as you try 0.1mm steps it will no longer reach the target. The Z axis seems to be even more problematic, because after just two 0.1mm steps the problem appears, whereas in XY slowly executing a few 0.1mm steps is possible.

I uploaded  the four changed files, so that you could test it out by yourself, if you have the time.
[https://github.com/DerM4209/Marlin_ECM_EDM_Mode/tree/main/Marlin_Test](url)

Thanks in advance :)

DerM4209, 2021-12-22:
Yes!,
I found another way to check if the machine reached the target by using:
`planner.has_blocks_queued()`

Now everything works as planned :)

DerM4209, 2022-01-08:
Here is the code that I've got so far:
https://github.com/DerM4209/Marlin_ECM_EDM_Mode
Can someone help me improving it, so that it can be integrated into the Marlin Firmware? 

Here is another video of it in use:
https://youtu.be/hA2yRkNJqhs

Cooper-8pTX, 2022-02-08:
@DerM4209 I also do ECM work and will try out your code in the coming weeks. I've found that marlin seems to have a lot of trouble executing really slow movements. I don't know if that will be a problem closed loop though.

DerM4209, 2022-03-09:
Hi @Cooper-8pTX ,
thank you for your interest.
Have you tried the code, yet?
I have almost finished the build of a Wire EDM machine, so that I can test it out for that. EDM Drilling / Sinking does work so far.
Sometimes it happened, that when the machine performed very small movements, the electrode did not move far enough away from the workpiece to clear the short, so that the machine could not continue in the program.
I solved that by bumping the electrode, but I think the code needs some changes to move further away than to the last position if the short can not be cleared by that.

Cooper-8pTX, 2022-03-09:
I've not had a chance to work on the firmware yet myself, but may in the coming month or two. I've found that Marlin doesn't like to really do much when the feed rate or distance are set too low. That may be part of your issue.

DerM4209, 2022-03-22:
In the last days I could make some progress with my Wire EDM machine.
https://youtu.be/JpsVs2Xm-9E
I think integration of Wire EDM (and also ECM) would be really useful for Marlin Users, because machines like the one that I built do barely cost more than 3D printers and you can precisely cut thick metal parts with it. So you can make all your *flat / extruded 2D*  parts out of metal instead of plastic  and you can also make 3D parts out of multiple metal pieces stacked on top of each other. It's also not as loud as CNC milling and so it's suited for apartments and working at night.

DerM4209, 2022-04-09:
Hi @Cooper-8pTX ,
besides Wire EDM I also want to try out ECM. I saw your Valentine-ECM-Mod and want to build it and test it out with the code for closed loop movement.

DerM4209, 2022-04-10:
@Cooper-8pTX ,
I spent the day building an ECM machine based on your Valentine-ECM-Mod to test it out with closed loop control.
It worked very well and I really like the way ECM works - It seems there is not much that can go wrong (like wire break with EDM) and by using closed loop control, the feedrate should never be too fast or too slow.
At my first try the metal removing rate was very low, but maybe I can improve it somehow in the future.

Here is a video of the test:
https://youtu.be/hf4My9hoJ5M

e7leef, 2023-01-01:
can you contact please, im interest in your ecm machine video on youtube and i really want help on it m9839452@gmail.com @DerM4209

DerM4209, 2023-01-04:
@e7leef 
I've sent you an email.

pas111954, 2025-07-31:
I own an old Jiten 25 old EDM discharging machine. This is my problem and wish someone can help:
Last Friday after more than 5 hours the EDM was working on a work piece I switched off the machine and went home. On Monday I restart the machine but the arm instead of going down to the workpiece it goes up until it stopped to the set program.
Like all is working in the control but the arm retract instead of going down to the workpiece piece.
Can anyone explain or tell me what went wrong and how to repair this issue
Thank you
Alvaro