MarlinFirmware/Marlin #22853: [BUG] probing gives different result after moving around the bed (delta)

LVD-AC, 2021-09-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The probe results are erratic; probing 4 points : centre and the 3 towers and immediatly probing the centre again gives about 1mm difference.

### Bug Timeline

somewhere between 1.1.9 and now

### Expected behavior

To give the same result no matter what the moves are in between probing the same point.

### Actual behavior

Gives different result at the same point after moving around.

### Steps to Reproduce

G28 ; home
G30 X0 Y0 ; probe the centre
G30 X60.62 Y-35 ; probe y-tower
G30 X0 Y+70 ; probe z-tower
G30 X60.62 Y-35 ; probe x-tower
G30 X0 Y0 ; probe the centre again

and compare the probe results of the centre point

2.0.x-bugfix

14:22:21.539 : X:0.00 Y:0.00 Z:258.84 E:0.00 Count A:45220B:45220C:45220
14:22:28.977 : Bed X: 0.00 Y: 0.00 Z: **-0.57**
14:22:28.977 : X:0.00 Y:0.00 Z:-0.87 E:0.00 Count A:19249B:19249C:19249
14:22:39.039 : Bed X: 60.62 Y: -35.00 Z: -0.14
14:22:39.039 : X:60.62 Y:-35.00 Z:-0.44 E:0.00 Count A:15918B:21539C:15917
14:22:52.882 : Bed X: 0.00 Y: 70.00 Z: -0.84
14:22:52.882 : X:0.00 Y:70.00 Z:-1.14 E:0.00 Count A:15825B:15835C:21469
14:23:02.194 : Bed X: -60.62 Y: -35.00 Z: 0.02
14:23:02.210 : X:-60.62 Y:-35.00 Z:-0.28 E:0.00 Count A:21555B:15924C:15913
14:23:08.803 : Bed X: 0.00 Y: 0.00 Z: **-1.50**
14:23:08.803 : X:0.00 Y:0.00 Z:-1.80 E:0.00 Count A:19156B:19156C:19156

1.1.9

14:33:22.591 : X:0.00 Y:0.00 Z:258.99 E:0.00 Count X:45234 Y:45234 Z:45234
14:33:30.359 : Bed X: 0.00 Y: 0.00 Z: **-0.16**
14:33:30.362 : X:0.00 Y:0.00 Z:-0.46 E:0.00 Count X:19290 Y:19290 Z:19290
14:33:40.310 : Bed X: 60.62 Y: -35.00 Z: -0.06
14:33:40.315 : X:60.62 Y:-35.00 Z:-0.36 E:0.00 Count X:15925 Y:21547 Z:15923
14:33:50.605 : Bed X: 0.00 Y: 70.00 Z: -0.19
14:33:50.610 : X:0.00 Y:70.00 Z:-0.49 E:0.00 Count X:15889 Y:15899 Z:21534
14:34:03.880 : Bed X: -60.62 Y: -35.00 Z: -0.19
14:34:03.884 : X:-60.62 Y:-35.00 Z:-0.49 E:0.00 Count X:21534 Y:15902 Z:15891
14:34:15.211 : Bed X: 0.00 Y: 0.00 Z: **-0.16**
14:34:15.211 : X:0.00 Y:0.00 Z:-0.46 E:0.00 Count X:19290 Y:19290 Z:19290

### Version of Marlin Firmware

2.0.x-bugfix

### Printer model

Flsun

LVD-AC, 2021-09-29:
Also notice that in 1.1.9 the bed is perceived flat at -0.15 +/- 0.05mm and prints a perfect 1st layer; and with exactly the same settings for endstop adjustments, delta radius, delta height and tower angle adjustments copied into 2.0.x-bugfix this is no longer the case. I'm pretty sure 1.1.9 reports the correct hights.

LVD-AC, 2021-10-05:
The last good version is **2.0.7.2** and the bug is still present in Bugfix-2.0.x even with PR #22795; the bug seems to get bigger with consecutive versions.

all tests with FLSUN autocalibrate configuration files and the exact same settings for the default delta parameters:

```
  // Distance between bed and nozzle Z home position
  #define DELTA_HEIGHT 259.14             // (mm) Get this value from G33 auto calibrate

  #define DELTA_ENDSTOP_ADJ { -1.53, -1.70, 0.0 } // Get these values from G33 auto calibrate

  // Horizontal distance bridged by diagonal push rods when effector is centered.
  #define DELTA_RADIUS 100.67              // (mm) Get this value from G33 auto calibrate

  // Trim adjustments for individual towers
  // tower angle corrections for X and Y tower / rotate XYZ so Z tower angle = 0
  // measured in degrees anticlockwise looking from above the printer
  #define DELTA_TOWER_ANGLE_TRIM { +0.17, -0.02, -0.15 } // Get these values from G33 auto calibrate

```

Tested 1.1.9.1: NO BUG

```
13:09:29.627 : X:0.00 Y:0.00 Z:259.04 E:0.00 Count X:45240 Y:45240 Z:45240
13:09:35.597 : **Bed X: 0.00 Y: 0.00 Z: -0.27**
13:09:35.601 : X:0.00 Y:0.00 Z:-0.37 E:0.00 Count X:19299 Y:19299 Z:19299
13:09:37.594 : Bed X: 56.00 Y: -33.00 Z: -0.18
13:09:37.598 : X:56.00 Y:-33.00 Z:-0.28 E:0.00 Count X:16329 Y:21478 Z:16266
13:09:40.445 : Bed X: 0.00 Y: 65.00 Z: -0.37
13:09:40.445 : X:0.00 Y:65.00 Z:-0.47 E:0.00 Count X:16259 Y:16268 Z:21459
13:09:46.803 : Bed X: -56.00 Y: -33.00 Z: -0.27
13:09:46.806 : X:-56.00 Y:-33.00 Z:-0.37 E:0.00 Count X:21469 Y:16311 Z:16239
13:09:50.707 : **Bed X: 0.00 Y: 0.00 Z: -0.28**
13:09:50.707 : X:0.00 Y:0.00 Z:-0.38 E:0.00 Count X:19298 Y:19298 Z:19298
```

Tested 2.0.0: NO BUG (but 0.06mm offset compared to 1.1.9.1)

```
14:11:18.365 : X:0.00 Y:0.00 Z:259.04 E:0.00 Count A:45240 B:45240 C:45240
14:11:21.977 : **Bed X: 0.00 Y: 0.00 Z: -0.33**
14:11:21.981 : X:0.00 Y:0.00 Z:-0.43 E:0.00 Count A:19293 B:19293 C:19293
14:11:23.446 : Bed X: 56.00 Y: -33.00 Z: -0.25
14:11:23.446 : X:56.00 Y:-33.00 Z:-0.35 E:0.00 Count A:16322 B:21471 C:16259
14:11:25.494 : Bed X: 0.00 Y: 65.00 Z: -0.42
14:11:25.498 : X:0.00 Y:65.00 Z:-0.52 E:0.00 Count A:16254 B:16263 C:21454
14:11:27.545 : Bed X: -56.00 Y: -33.00 Z: -0.34
14:11:27.545 : X:-56.00 Y:-33.00 Z:-0.44 E:0.00 Count A:21462 B:16304 C:16232
14:11:29.010 : **Bed X: 0.00 Y: 0.00 Z: -0.33**
14:11:29.014 : X:0.00 Y:0.00 Z:-0.43 E:0.00 Count A:19293 B:19293 C:19293
```

Tested 2.0.1: NO BUG (but 0.07mm offset compared to 1.1.9.1)

```
14:03:13.076 : X:0.00 Y:0.00 Z:259.04 E:0.00 Count A:45240 B:45240 C:45240
14:03:16.692 : **Bed X: 0.00 Y: 0.00 Z: -0.33**
14:03:16.702 : X:0.00 Y:0.00 Z:-0.43 E:0.00 Count A:19293 B:19293 C:19293
14:03:18.157 : Bed X: 56.00 Y: -33.00 Z: -0.27
14:03:18.160 : X:56.00 Y:-33.00 Z:-0.37 E:0.00 Count A:16320 B:21469 C:16257
14:03:20.207 : Bed X: 0.00 Y: 65.00 Z: -0.42
14:03:20.211 : X:0.00 Y:65.00 Z:-0.52 E:0.00 Count A:16254 B:16263 C:21454
14:03:22.258 : Bed X: -56.00 Y: -33.00 Z: -0.34
14:03:22.259 : X:-56.00 Y:-33.00 Z:-0.44 E:0.00 Count A:21462 B:16304 C:16232
14:03:23.729 : **Bed X: 0.00 Y: 0.00 Z: -0.35**
14:03:23.729 : X:0.00 Y:0.00 Z:-0.45 E:0.00 Count A:19291 B:19291 C:19291
```

Tested 2.0.3: NO BUG (but 0.07mm offset compared to 1.1.9.1)

[... 100 lines cut ...]
11:39:14.970 : Bed X: 56.00 Y: -33.00 Z: -0.22
11:39:14.975 : X:56.00 Y:-33.00 Z:-0.32 E:0.00 Count A:16325B:21474C:16262
11:39:36.764 : Bed X: 0.00 Y: 65.00 Z: -0.89
11:39:36.769 : X:0.00 Y:65.00 Z:-0.99 E:0.00 Count A:16207B:16216C:21407
11:39:44.831 : Bed X: -56.00 Y: -33.00 Z: -0.00
11:39:44.834 : X:-56.00 Y:-33.00 Z:-0.10 E:0.00 Count A:21496B:16338C:16266
11:39:50.779 : **Bed X: 0.00 Y: 0.00 Z: -1.28**
11:39:50.783 : X:0.00 Y:0.00 Z:-1.38 E:0.00 Count A:19198B:19198C:19198
```

Tested Bugfix-2.0.x with PR #22795 : bug present (0.90mm difference at origin)

```
13:44:24.070 : X:0.00 Y:0.00 Z:259.04 E:0.00 Count A:45240B:45240C:45240
13:44:27.723 : **Bed X: 0.00 Y: 0.00 Z: -0.59**
13:44:27.723 : X:0.00 Y:0.00 Z:-0.69 E:0.00 Count A:19267B:19267C:19267
13:44:29.184 : Bed X: 56.00 Y: -33.00 Z: -0.13
13:44:29.188 : X:56.00 Y:-33.00 Z:-0.23 E:0.00 Count A:16334B:21483C:16271
13:44:31.282 : Bed X: 0.00 Y: 65.00 Z: -1.06
13:44:31.282 : X:0.00 Y:65.00 Z:-1.16 E:0.00 Count A:16190B:16199C:21390
13:44:33.295 : Bed X: -56.00 Y: -33.00 Z: 0.38
13:44:33.298 : X:-56.00 Y:-33.00 Z:0.28 E:0.00 Count A:21534B:16376C:16304
13:44:36.336 : **Bed X: 0.00 Y: 0.00 Z: -1.49**
13:44:36.336 : X:0.00 Y:0.00 Z:-1.59 E:0.00 Count A:19177B:19177C:19177
```

LVD-AC, 2021-12-05:
This problem still exists

dellipse, 2022-01-03:
Problem still exists.

LVD-AC, 2022-03-05:
still no change

closed 2022-05-17 by github-actions[bot]