MarlinFirmware/Marlin #22855: [BUG] BLTouch doesnt home

flynhigh20, 2021-09-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Zprobe is eother in triggered state on boot or open but crashes in bed, if its triggered i can home it but cant lower z axis

### Expected behavior

Probe to deploy and state triggered.

### Actual behavior

Not triggering or its reversed bit marlin wont let be invert probe pins.

### Version of Marlin Firmware

2.0.09.1

### Printer model

Cr10v2 heavily modded

### Electronics

Bigtreetech octopus v1.1

### Add-ons

MMU2S

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

flynhigh20, 2021-09-29:
Did work on creality crx motherboard

jonhysfl, 2021-09-29:
Same issue here in Anet A8 Plus.
This line:

#define Z_MIN_PROBE_ENDSTOP_INVERTING false  // Set to true to invert the logic of the probe.

cannot be set to true

ellensp, 2021-09-29:
The only way you can have Z_MIN_PROBE_ENDSTOP_INVERTING set wrong is if you not running current code.
In which case you have failed to try it on buxfix.

```
#if ENABLED(BLTOUCH)
  // Always disable probe pin inverting for BLTouch
  #if Z_MIN_PROBE_ENDSTOP_INVERTING
    #error "BLTOUCH requires Z_MIN_PROBE_ENDSTOP_INVERTING set to false. Please update your Configuration.h file."
  #endif

```

ellensp, 2021-09-29:
You have failed to provide any relevant details.
1) what bltouch options you enabled to tell marlin where you plugged the bltouch into
2) where you physically plugged in the bltouch

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-09-29 by ellensp