MarlinFirmware/Marlin #22856: [FR] Add support of Alfawise Ex8 LCD. (ST7565_64128N SPI)

vmilevski, 2021-09-30:
### Describe the feature you want

I need to connect LCD from Alfawise Ex8 3d printer marked as WYH L12864 COG (see attachment). This is ST7565_64128N based board. I get this LCD worked as AZSMZ_12864 (#define AZSMZ_12864 in the Configuration.h) but there is picture offset for 4 pixels from right to left. See attachment. Others ST7565 lcds for example MAKRPANEL or VIKI2 also works with same result. Where is a place to correct this problem and add support for this LCD to Marlin?  
[image]

ellensp, 2021-09-30:
Some hints can be found here https://3dprinting.stackexchange.com/questions/11145/anet-a8-clone-lcd-not-working-after-flash 
But for older marlin v1

ellensp, 2021-09-30:
created a diff from the linked modified marlin 1 files

```DIFF
diff --git a/Marlin/ultralcd_st7565_u8glib_VIKI.h b/Marlin/ultralcd_st7565_u8glib_VIKI.h
index adb48ea6b2..8efca8ed3b 100644
--- a/Marlin/ultralcd_st7565_u8glib_VIKI.h
+++ b/Marlin/ultralcd_st7565_u8glib_VIKI.h
@@ -144,6 +144,7 @@
 #define ST7565_NOOP()            ST7565_WRITE_BYTE(0xE3)
 
 uint8_t u8g_dev_st7565_64128n_2x_VIKI_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg) {
+	uint8_t i, y;		   
   switch (msg) {
 
     case U8G_DEV_MSG_INIT: {
@@ -183,7 +184,21 @@ uint8_t u8g_dev_st7565_64128n_2x_VIKI_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg
 
       ST7565_INVERTED(0);               // display normal, bit val 0: LCD pixel off.
 
-      ST7565_CONTRAST(0x1E);            // Contrast value. Setting for controlling contrast of Displaytech 64128N
+      ST7565_CONTRAST(0x38);            // Contrast value. Setting for controlling contrast of Displaytech 64128N
+
+      for (y = 0; y < 9; y++) { //clear GDRAM
+        ST7565_WRITE_BYTE(0x0B0+y);   //set page-8
+        ST7565_WRITE_BYTE(0x010);     //set column
+        ST7565_WRITE_BYTE(0x000);     //set column
+        ST7565_A0();                     /* instruction mode */
+        for (i = 0; i < 132; i++)     //
+          ST7565_WRITE_BYTE(0);
+        ST7565_NA0();                     /* instruction mode */
+      }
+
+
+      ST7565_NA0();   
+
 
       ST7565_ON(1);                     // display on
 
@@ -203,7 +218,7 @@ uint8_t u8g_dev_st7565_64128n_2x_VIKI_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg
       ST7565_CS();                      // chip select off
       ST7565_NA0();                     // instruction mode
       ST7565_NCS();                     // chip select
-      ST7565_COLUMN_ADR(0x00);          // high 4 bits to 0, low 4 bits to 0. Changed for DisplayTech 64128N
+      ST7565_COLUMN_ADR(0x00+4);          // high 4 bits to 0, low 4 bits to 0. Changed for DisplayTech 64128N
                                         // end of sequence
       ST7565_PAGE_ADR(2 * pb->p.page);  // select current page (ST7565R)
       ST7565_A0();                      // data mode
@@ -211,7 +226,7 @@ uint8_t u8g_dev_st7565_64128n_2x_VIKI_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg
       ST7565_CS();                      // chip select off
       ST7565_NA0();                     // instruction mode
       ST7565_NCS();                     // chip select
-      ST7565_COLUMN_ADR(0x00);          // high 4 bits to 0, low 4 bits to 0
+      ST7565_COLUMN_ADR(0x00+4);          // high 4 bits to 0, low 4 bits to 0
                                         // end of sequence
       ST7565_PAGE_ADR(2 * pb->p.page + 1); // select current page (ST7565R)
       ST7565_A0();                      // data mode
@@ -224,6 +239,7 @@ uint8_t u8g_dev_st7565_64128n_2x_VIKI_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg
       ST7565_NCS();
       ST7565_NA0();                     // instruction mode
       ST7565_CONTRAST((*(uint8_t*)arg) >> 2);
+		ST7565_CONTRAST(0x38);					
       ST7565_CS();                      // chip select off
       return 1;
 
@@ -265,4 +281,4 @@ class U8GLIB_ST7565_64128n_2x_VIKI : public U8GLIB {
 
 #pragma GCC reset_options
 
-#endif // ULCDST7565_H
+#endif // ULCDST7565_H
\ No newline at end of file

```

The important change seems to be 

ST7565_COLUMN_ADR(0x00+4)

now to find how to implement this in current marlin...

ellensp, 2021-09-30:
Try this (this is an educated guess, since I don't have the hardware) 

Build marlin, this populates the .pio sub directory

In Marlin/.pio/libdeps/{ your board name }/U8glib-HAL/src/clib/u8g_dev_st7565_64128n.c
if the function
```CPP
static const uint8_t u8g_dev_st7565_64128n_data_start[] PROGMEM = {
  U8G_ESC_ADR(0),           /* instruction mode */
  U8G_ESC_CS(1),            /* enable chip */
  0x010,                        /* set upper 4 bit of the col adr to 0x10 */
  0x000,                            /* set lower 4 bit of the col adr to 0x00. Changed for DisplayTech 64128N */
  U8G_ESC_END               /* end of sequence */
};
```
change the 0x000 to 0x004 and recompile

NB this the u8glib-hal  library that is pulled in, if you hit build clean it will delete this file and get a fresh copy.
So don't clean before recompiling.

vmilevski, 2021-09-30:
> ST7565_COLUMN_ADR(0x00+4)
Thanks a lot! This is an answer!
It is needed to change make this change in the Marlin/src/lcd/dogm/u8g_dev_st7565_64128n_HAL.cpp
I added
#ifndef XOFFSET
  #define XOFFSET 0x00
#endif

at the begiiing of the file and then change main function to 

static const uint8_t u8g_dev_st7565_64128n_HAL_data_start[] PROGMEM = {
  U8G_ESC_ADR(0),             // instruction mode
  U8G_ESC_CS(1),              // enable chip
  
  ST7565_COLUMN_ADR(XOFFSET), // high 4 bits to 0, low 4 bits to 0. Changed for DisplayTech 64128N
  
  U8G_ESC_END                 // end of sequence
};
And set this variable in the Marlin/src/inc/Conditionals_LCD.h 
  #elif ENABLED(WYH_L12864)
    #define U8GLIB_ST7565_64128N
    #define XOFFSET 0x04

 WYH_L12864 - define name which I choose for this display.
 Once again thanks a lot. 
 PS Is it needed to make change request or this is not needed die to any reason?

ellensp, 2021-09-30:
You should make a PR requests to add this permanently into marlin and u8glib-hal library  (or I can do it.)

VragVideo, 2021-10-01:
> You should make a PR request to add this permanently into marlin (or I can do it.)

Just create change request under my personal account (you can compare email basis). Once again thank for your quick and concrete help. If there will be problems with my change request create you own please.

ellensp, 2021-10-02:
Since PR has been created and merge is imminent. We can close this

closed 2021-10-02 by ellensp