MarlinFirmware/Marlin #22857: [FR] (feature summary)

eric-s-raymond, 2021-09-30:
### Is your feature request related to a problem? Please describe.

Main screen of Ender 3V2 console interface is difficult to use under bright light because the contrast between the reverse-video selection highlifhght and the display background is too low - I find I have to shield the display with a hand or peer at it very closely for the selection highlight to be visible.

This could be due to a poor choice of hardware or settings by Creality and might not replicate onb other printers, but the firmware ought to be designed for robustness of UI in the worst case.

### Describe the feature you want

More prominent visual indication of which item is selected - an obvious color change or a box outline or something of the sort.

The-EG, 2021-09-30:
There are now 4 different UI variants available on the Ender 3 v2. You'll need to specify which one you are using at the very least.

eric-s-raymond, 2021-09-30:
I'm usinbg the non-touchscreen variant of the BLtouch firmware - .  I observed the same behavior on non-BLtouch stock 3V2 firmware.

My console no longer has an info screen (I should report thsat as a bug) but it's what the Creality downlods page lists as "26 Mar, 2021  Ender-3 V2-Marlin-2.0.1-HW-4.2.2-mainboard-V1.1.2-Compatible with BLTouch and filament detection"

thisiskeithb, 2021-09-30:
We can’t support forks of Marlin here, so please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code before opening new feature requests or bug reports.

ellensp, 2021-09-30:
Closing, can reopen if issue is found to also exists in actual Marlin

closed 2021-09-30 by ellensp