MarlinFirmware/Marlin #22858: [FR] Action selections need positive visual acknowledgement

eric-s-raymond, 2021-09-30:
### Is your feature request related to a problem? Please describe.

Some console menu options do not provide any visual cue that the firmware has registered the request, leading to user uncertainty about the state of the machine and what to do next.

If a choice results in goiing to a different menu, or in a visible action such as autohoming or bed leveling, this is not a problem.  But for some functions, like disabling stepper moters, the interface gives no feedback.

### Describe the feature you want

Every menu selection that does not jump to another menu should produce a clear visual acknowledgement.  Something like painting a checkmark near the right end of the selected area, or a color change of the highlight bar.

This is pretty basic for good UI design.

The-EG, 2021-09-30:
same as #22857, which of the 4 UIs are you using on the Ender 3 v2?

eric-s-raymond, 2021-09-30:
I'm using the non-touchscreen variant of the BLtouch firmware - .  I observed the same behavior on non-BLtouch stock 3V2 firmware.

My console no longer has an info screen (I should report thsat as a bug) but it's what the Creality downloads page lists as "26 Mar, 2021  Ender-3 V2-Marlin-2.0.1-HW-4.2.2-mainboard-V1.1.2-Compatible with BLTouch and filament detection"

I strongly suspect this UI bug is common to all versions.

thisiskeithb, 2021-09-30:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22857

eric-s-raymond, 2021-09-30:
No, this is *not* a duplicate of #22587.   These are different UI bugs that happen with different preconditions in different places. #22857 is a UI problem with the interior menus and manifests after the user has attempted an action; this is a problem with tile selection that manifests *before* the user attempts an action.