MarlinFirmware/Marlin #22860: filament change after x meter of filament

rooiejoris, 2021-09-30:
### Is your feature request related to a problem? Please describe.

When printing a part larger than the material on a spool then it would be nice to 'program' a filament change in the future, say after x meter.
This could be usefull when printing a part larger than a standard spool of 100/65/x meters, but also when you realise that the material on the real is going to end before the print is finished.
Many times you think: I will check again in 5 minutes, but in real live you forget because of a phone call or other some other distractions.
Same option could be at pause print after layer x or height y or filament length z.

And yes, I am aware that this functionality could also be solved in Octoprint or the slicer, but let's discuss what is the best place... : )

### Describe the feature you want

Maybe a solution is to extend the M600 command with a parameter to set 'after x meters'...(?!) This would work for an octoprint like setup where you can send custom commands.
Or extend the display option of a filament change with the option 'after x meters' but it should default to 0 so current behaviour is almost not changed.

ellensp, 2021-09-30:
This should be a slicer feature, not firmware

rooiejoris, 2021-10-01:
Can we agree on both... : )

Like all the setting in tune menu are slicing features (no?!).
But in real life it is very useful to adjust them during the print since you cannot think of everything in advance and it is really helpful to adjust them during the print. Especially with larger prints.

Case 1: preparing for multiple small spools I completely agree I want to set that in the slicer. That's sort of how I did it last times, I adjusted the gcode manually.

Case 2:  I started the print, but after >12 hours you realize there is just not enough material and you see x meters left on the spool. You can do a material change, but it is nice to set the change after x meters to finish even more material.

Case 3: You start the print and realize you can print the first part much faster but you are sure that the top part should be a lot slower. A 'delayed' speed adjustment could be helpful. Also the other way round, when printing a 'more difficult' material, you realize that the first layer has to be printed way slower than expected, but you are sure that after 10 layers the speed can go up.

Case 4: .....

It all depends on how you design you workflow. For me I do a lot of thing on the fly based on the feedback I get from the print and the printer.

(and yes the same discussion can be made at the octoprint and slicer forums...)

InsanityAutomation, 2021-10-01:
If you enable a runout motion sensor even without one physically hooked up and use M412D1000S1 to set it to 1 meter for example you can sortve create this functionality with what is present now...

rooiejoris, 2021-10-01:
That is a really nice workaround I think...!!
Have to dive into the M412 gcode command.

THANKS....!!

however It doesn't solve case 3....  : )

InsanityAutomation, 2021-10-01:
I see no way to really trigger a speed change consistently based on some future prediction of position. That one is well outside the scope of what I'd consider firmware to be expected to do.  The other while a stretch, you can manipulate existing features to do so. Is it really correct / design intent? No. Will it work? Sure.

ellensp, 2021-10-02:
Basically i'm of the opinion if it can be done in a slicer, there is no need for it to be in the firmware. 
Firmware resources are limited, in comparison the slicers on a pc that has near infinite resources....

rooiejoris, 2021-10-03:
I agree about the limited resources, although the way I work with the printers there is a need to have it in the firmware... : )
But there are workarounds, so I am not pushing too much here...