MarlinFirmware/Marlin #22861: [FR] (Change # of Mesh Grid Points on TFT screen)

andrewjbuehler, 2021-09-30:
### Describe the feature you want

I use a 7 x 7 mesh grid when doing big prints but it takes a long time to complete. It would be nice to have an easy way to change the number of grid points from my printer screen so I can scale it down easily for a faster print start up.

ellensp, 2021-09-30:
Many tft screen are actually small computers running their own firmware. The supplers of that firmware would need to update thier firmware.

InsanityAutomation, 2021-10-01:
Another issue with attempting to do so is the change in grid size would push a notable amount of data to RAM, which is extremely limited on most boards in use (8kb for a 2560 MCU). Most often the grid size is limited in order to fit within the memory limits, and if it was dynamically configurable you would need to size the mesh arrays in RAM and eeprom for the largest possible grid.

Best practice is to probe a solid mesh once, and store it to EEPROM. If the machine is well adjusted mechanically, it wont change until some event occurs to change it. Ive recalled the same mesh for months on several machines without redoing the probing sequence.

thisiskeithb, 2022-04-09:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/10547

closed 2022-04-09 by thisiskeithb