MarlinFirmware/Marlin #22862: [BUG] The third Z-axis motor is not activated.

alxxxx, 2021-10-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have set the third Z axis for "Z Steppers Auto-Alignment".
Registered in the configs the number of Z motors and the driver for the third Z motor.
The Z1 and Z2 motors move, but the Z3 motor does not.
I checked the motor and driver, everything is fine with them.
If you go to the "Driver Current" menu, the Z3 motor is missing.
At the same time, if you move the extruder motor, the Z3 motor will move !!!!
And the extruder motor does not work at all !!!
As far as I understand, when assembling the firmware for the BTT Octopus board, the NUM_Z_STEPPER_DRIVERS parameter is not taken into account.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7266466/configs.zip)

### Version of Marlin Firmware

2.0.9.1

### Printer model

BLV Cube

### Electronics

BTT Octopus V1.1 + BTT TFT50

### Bed Leveling

ABL Bilinear mesh

ellensp, 2021-10-01:
Provided configs are not bugfix configs, you you really test this on bugfix?

ellensp, 2021-10-01:
[Bugfix Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7266648/Bugfix.Configuration.zip)
For testing

alxxxx, 2021-10-01:
> 
> 
> Provided configs are not bugfix configs, Did you really test this on bugfix?

Yes, I actually checked the bugfix firmware
[Configs bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/7266676/Configs.bugfix.zip)
.

alxxxx, 2021-10-01:
> 
> 
> [Bugfix Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7266648/Bugfix.Configuration.zip) For testing
Your files are virtually indistinguishable from mine. For some reason you reduced Z_MAX_POS and some other insignificant trifle.

alxxxx, 2021-10-01:
I temporarily solved the problem by correcting the pins_BTT_OCTOPUS_V1_common.h file
Added pins for Z3, E0 moved by 1 position.

I do not know how much this is correct, but all 3 motors and the extruder motor are working.
Shouldn't this happen automatically when compiling the firmware?

```
#define X_DIAG_PIN                          PG6   // X-STOP
#define Y_DIAG_PIN                          PG9   // Y-STOP
#define Z_DIAG_PIN                          PG10  // Z-STOP
#define Z2_DIAG_PIN                         PG11  // Z2-STOP
#define Z3_DIAG_PIN                         PG12  // Z3-STOP
#define E0_DIAG_PIN                         PG13  // E0DET
#define E1_DIAG_PIN                         PG14  // E1DET
#define E2_DIAG_PIN                         PG15  // E2DET

....

#define FIL_RUNOUT_PIN                      PG13  // E0DET
#define FIL_RUNOUT2_PIN                     PG14  // E1DET
#define FIL_RUNOUT3_PIN                     PG15  // E2DET

....

#define Z3_STEP_PIN                         PF9   // MOTOR 4    Z-Axe 3
#define Z3_DIR_PIN                          PF10
#define Z3_ENABLE_PIN                       PG2
#ifndef Z3_CS_PIN
  #define Z3_CS_PIN                         PF2
#endif

#define E0_STEP_PIN                         PC13  // MOTOR 5   Extruder 0
#define E0_DIR_PIN                          PF0
#define E0_ENABLE_PIN                       PF1
#ifndef E0_CS_PIN
  #define E0_CS_PIN                         PE4
#endif

#define E1_STEP_PIN                         PE2   // MOTOR 6
#define E1_DIR_PIN                          PE3
#define E1_ENABLE_PIN                       PD4
#ifndef E1_CS_PIN
  #define E1_CS_PIN                         PE1
#endif

#define E2_STEP_PIN                         PE6   // MOTOR 7
#define E2_DIR_PIN                          PA14
#define E2_ENABLE_PIN                       PE0
#ifndef E2_CS_PIN
  #define E2_CS_PIN                         PD3
#endif

```

ellensp, 2021-10-01:
Z steper 
 > PIN: PF11        M42 P91          Z_STEP_PIN                             protected 
 > PIN: PG3         M42 P99          Z_DIR_PIN                              protected 
 > PIN: PG5         M42 P101         Z_ENABLE_PIN                           protected 
 > PIN: PC6         M42 P38          Z_SERIAL_TX_PIN                        Output = 1
 > .                                 Z_SERIAL_RX_PIN                        Output = 1
 > PIN: PG10        M42 P106         Z_DIAG_PIN                             Input  = 1
 > .                                 Z_STOP_PIN                             Input  = 1

 Z2 steper 

 > PIN: PG4         M42 P100         Z2_STEP_PIN                            protected 
 > PIN: PC1         M42 P33          Z2_DIR_PIN                             protected 
 > PIN: PA0         M42 P0           Z2_ENABLE_PIN                          protected 
 > .                                 Z2_SERIAL_TX_PIN                       Output = 1
 > .                                 Z2_SERIAL_RX_PIN                       Output = 1
 > PIN: PC7         M42 P39          Z2_CS_PIN                              Output = 1

 Z3 steper 
 > PIN: PC13        M42 P45          E1_STEP_PIN                            protected 
 > .                                 Z3_STEP_PIN                            protected 
 > PIN: PF0         M42 P80          E1_DIR_PIN                             protected 
 > .                                 Z3_DIR_PIN                             protected 
 > PIN: PF1         M42 P81          E1_ENABLE_PIN                          protected 
 > .                                 Z3_ENABLE_PIN                          protected 
 > PIN: PE4         M42 P68          E1_CS_PIN                              Output = 1
 > .                                 Z3_SERIAL_TX_PIN                       Output = 1
 > .                                 Z3_SERIAL_RX_PIN                       Output = 1
 > .                                 E1_SERIAL_TX_PIN                       Output = 1
 > .                                 E1_SERIAL_RX_PIN                       Output = 1
 > PIN: PE4         M42 P68          E1_CS_PIN                              Output = 1

ellensp, 2021-10-01:
I think your using the wrong stepper plug

Z is predefined in pins, so is not changed
Z2 is predefined in pins so is not changed
Z3 is not predefined so marlin uses the first available unused E stepper in this case E1 Ie it uses motor 5 as Z3

alxxxx, 2021-10-01:
> 
> 
> I think your using the wrong stepper plug
> 
> Z is predefined in pins, so is not changed, uses MOTOR 2 Z2 is predefined in pins so is not changed, uses MOTOR 3 E0 is predefined in pins so is not changed, uses MOTOR 4 Z3 is not predefined so marlin uses the first available unused E stepper in this case E1 Ie it uses MOTOR 5 as Z3

A funny nuance, but if it were as you write, then instead of the Z3 motor, the E0 motor would work for me.
But I'll check now and let you know.
And yes .. in the printer menu "Driver Current" 1 more motor has not appeared, only X Y Z1 Z2 E0.

alxxxx, 2021-10-01:
So. I have reverted the pins_BTT_OCTOPUS_V1_common.h file to its original state.
And did as you said, turned on the extruder in MOTOR 4 (E0), and Z3 in MOTOR 5 (E1).
As a result, the extruder motor works.
Motor Z3 - no.
Those. The firmware does not see that there are 3 Z-axis motors in the config and will not redefine the MOTOR 5 (E1) connector.

[M43 output.txt](https://github.com/MarlinFirmware/Marlin/files/7267115/M43.output.txt)

ellensp, 2021-10-01:
You are NOT running bugfix!

ellensp, 2021-10-01:
Thus you are triggering an already fixed bug, where Z2 ends up on E2 instead of E1 

from your M43 output
```
15:13:41.031 : PIN: PD4         M42 P52          E2_ENABLE_PIN                          protected
15:13:41.031 : .                                 Z3_ENABLE_PIN                          protected
15:13:41.031 : PIN: PE1         M42 P65          E2_CS_PIN                              Output = 1
15:13:41.031 : .                                 Z3_SERIAL_TX_PIN                       Output = 1
15:13:41.031 : .                                 Z3_SERIAL_RX_PIN                       Output = 1
15:13:41.046 : .                                 E2_SERIAL_TX_PIN                       Output = 1
15:13:41.046 : .                                 E2_SERIAL_RX_PIN                       Output = 1
15:13:41.046 : PIN: PE2         M42 P66          E2_STEP_PIN                            protected
15:13:41.046 : .                                 Z3_STEP_PIN                            protected
15:13:41.046 : PIN: PE3         M42 P67          E2_DIR_PIN                             protected
15:13:41.046 : .                                 Z3_DIR_PIN                             protected

```
Ie Z2 ended up on E2 when it should have ended up on E1

ellensp, 2021-10-01:
For reference, was fixed here, 17 days ago. https://github.com/MarlinFirmware/Marlin/pull/22771

closed 2021-10-01 by ellensp

alxxxx, 2021-10-01:
> 
> 
> Thus you are triggering an already fixed bug, where Z3 ends up on E2 instead of E1
> 
> from your M43 output
> 
> ```
> 15:13:41.031 : PIN: PD4         M42 P52          E2_ENABLE_PIN                          protected
> 15:13:41.031 : .                                 Z3_ENABLE_PIN                          protected
> 15:13:41.031 : PIN: PE1         M42 P65          E2_CS_PIN                              Output = 1
> 15:13:41.031 : .                                 Z3_SERIAL_TX_PIN                       Output = 1
> 15:13:41.031 : .                                 Z3_SERIAL_RX_PIN                       Output = 1
> 15:13:41.046 : .                                 E2_SERIAL_TX_PIN                       Output = 1
> 15:13:41.046 : .                                 E2_SERIAL_RX_PIN                       Output = 1
> 15:13:41.046 : PIN: PE2         M42 P66          E2_STEP_PIN                            protected
> 15:13:41.046 : .                                 Z3_STEP_PIN                            protected
> 15:13:41.046 : PIN: PE3         M42 P67          E2_DIR_PIN                             protected
> 15:13:41.046 : .                                 Z3_DIR_PIN                             protected
> ```
> 
> Ie Z3 ended up on E2 when it should have ended up on E1

Before writing here, I CHECKED the firmware bugfix !!!!!
On the bugfix, the problem is the same!
But I did not say anywhere that I am now testing on the bugfix firmware!

In general, what the hell is not the next free connector, but the MOTOR 6 connector!

I moved the driver to MOTOR 6 and plugged in the Z3 engine. Now it works.

This is extremely NOT LOGICAL. Why not just shift the row by the number of motors added?

ellensp, 2021-10-01:
This is simply not true.
Using you config files updated to bugfix code. On my Octopus On current bugfix produces this M43 output
[pins.raw.txt](https://github.com/MarlinFirmware/Marlin/files/7267297/pins.raw.txt)
Where Z3 ends up on E1 as it should.

InsanityAutomation, 2021-10-01:
> 
> 
> You are NOT running bugfix!

Most likely we will see many who think they were based on the GitHub web UI download link issue that was just fixed a few days ago...