MarlinFirmware/Marlin #22864: [BUG] Dual endstops Z. Will not compile

liamw8lde, 2021-10-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a Mpcnc Lowrider 2 and I am using dual Y and Z steppers with dual endstops. home to Z_max 
When I define the Z_MULTI_ENDSTOPS in the configuration_adv.h it will not compile anymore. 
I have defined in the configuration.h 

#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
#define USE_Y2MAX_PLUG
#define USE_Z2MAX_PLUG
#define USE_Y2MIN_PLUG
#define USE_Z2MIN_PLUG

#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
  // Enable if Z motor direction signals are the opposite of Z1
  #define INVERT_Z2_VS_Z_DIR
  //#define INVERT_Z3_VS_Z_DIR
  //#define INVERT_Z4_VS_Z_DIR

  #define Z_MULTI_ENDSTOPS
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _XMAX_
    #define Z2_ENDSTOP_ADJUSTMENT   0
 

### Bug Timeline

as of 30_09.21

### Expected behavior

enable 

### Actual behavior

will not compile.
                 from Marlin\src\module\endstops.cpp:27:
Marlin\src\module\endstops.cpp: In static member function 'static void Endstops::update()':
Marlin\src\module\endstops.cpp:703:32: error: 'Z2_MIN' was not declared in this scope; did you mean 'Y2_MIN'?
  703 |         COPY_LIVE_STATE(Z_MIN, Z2_MIN);

in enstops.h 
_ES_ITEM(HAS_Y2_MAX, Y2_MAX) has no value

### Steps to Reproduce

uncomment #define Z_MULTI_ENDSTOPS
add Y2 and X2 endstops to configuration.h 

### Version of Marlin Firmware

define CONFIGURATION_ADV_H_VERSION 02000902

### Printer model

Mpcnc Lowrider 2 

### Electronics

bigtreetech octopus

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Additional information & file uploads

[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/7270019/Marlin-bugfix-2.0.x.zip)

ellensp, 2021-10-02:
These are made up and do not do anything...
#define USE_Y2MAX_PLUG
#define USE_Z2MAX_PLUG
#define USE_Y2MIN_PLUG
#define USE_Z2MIN_PLUG

ellensp, 2021-10-02:
The issue is the auto assignment of endstop pins in Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h
With 2 Z endstops, you allocate a endstop plugs to use. But the pin file only creates under other conditions.

This will let you compile. but is not a complete solution.  (Ie it needs work)
```DIFF
diff --git a/Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h b/Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h
index ab0a7dd962..7e391ea7c1 100644
--- a/Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h
+++ b/Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h
@@ -72,7 +72,7 @@
   #else
     #define X_MIN_PIN                E0_DIAG_PIN  // E0DET
   #endif
-#elif EITHER(X_DUAL_ENDSTOPS, DUAL_X_CARRIAGE)
+#elif ANY(X_DUAL_ENDSTOPS, DUAL_X_CARRIAGE, Z_MULTI_ENDSTOPS)
   #ifndef X_MIN_PIN
     #define X_MIN_PIN                 X_DIAG_PIN  // X-STOP
   #endif
@@ -90,7 +90,7 @@
   #else
     #define Y_MIN_PIN                E1_DIAG_PIN  // E1DET
   #endif
-#elif ENABLED(Y_DUAL_ENDSTOPS)
+#elif EITHER(Y_DUAL_ENDSTOPS, Z_MULTI_ENDSTOPS)
   #ifndef Y_MIN_PIN
     #define Y_MIN_PIN                 Y_DIAG_PIN  // Y-STOP
   #endif
@@ -108,7 +108,7 @@
   #else
     #define Z_MIN_PIN                E2_DIAG_PIN  // PWRDET
   #endif
-#elif ENABLED(Z_MULTI_ENDSTOPS)
+#elif ENABLED(Z_MULTI_ENDSTOPS) && (Z2_USE_ENDSTOP == _ZMAX_|| Z2_USE_ENDSTOP == _ZMIN_)
   #ifndef Z_MIN_PIN
     #define Z_MIN_PIN                 Z_DIAG_PIN  // Z-STOP
   #endif

```

liamw8lde, 2021-10-02:
thankyou it complied, Ill be heading to the workshop tomorrow to install the new board in the Cnc and test it..

ellensp, 2021-10-04:
Created a PR with a much nicer patch to fix this properly.

ellensp, 2021-10-17:
PR's have been merged. Closing

closed 2021-10-17 by ellensp