MarlinFirmware/Marlin #22876: [BUG] Cannot compile the latest Bugfix or Stable

FlailAway, 2021-10-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I downloaded "Stable" and got this error without changing anything in Config etc. I then downloaded BF -- more of the same 'ol, same'ol.

` Arduino: 1.8.13 (Linux), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"
In file included from sketch/src/lcd/extui/ftdi_eve_touch_ui/language/language.h:23:0,
                 from sketch/src/lcd/extui/ftdi_eve_touch_ui/language/language.cpp:25:
sketch/src/lcd/extui/ftdi_eve_touch_ui/language/language_en.h:148:24: error: redefinition of 'const char Language_en::MSG_MESH_VIEW []'
   PROGMEM Language_Str MSG_MESH_VIEW                = u8"View Mesh";
                        ^~~~~~~~~~~~~
In file included from sketch/src/lcd/extui/ftdi_eve_touch_ui/language/../../../../inc/../core/language.h:563:0,
                 from sketch/src/lcd/extui/ftdi_eve_touch_ui/language/../../../../inc/MarlinConfig.h:53,
                 from sketch/src/lcd/extui/ftdi_eve_touch_ui/language/../../../../MarlinCore.h:24,
                 from sketch/src/lcd/extui/ftdi_eve_touch_ui/language/language.cpp:23:
sketch/src/lcd/extui/ftdi_eve_touch_ui/language/../../../../inc/../core/../lcd/language/language_en.h:160:8: note: 'const char Language_en::MSG_MESH_VIEW [10]' previously defined here
   LSTR MSG_MESH_VIEW                      = _UxGT("View Mesh");
        ^~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
`

### Bug Timeline

Fresh download today and compile

### Expected behavior

Compile without error

### Actual behavior

Would not compile

### Steps to Reproduce

Download
Extract
Load into Arduino IDE
Click "Compile"

### Version of Marlin Firmware

2.0.9.02 bf

### Printer model

Mine

### Electronics

BigtreeTech

### Bed Leveling

ABL Linear grid

### Your Slicer

IdeaMaker

### Additional information & file uploads

None

ellensp, 2021-10-03:
Yet another arduino IDE issue...  Compiles fine on vscode/platformio

ellensp, 2021-10-03:
Add a #if ENABLED block eg 

```DIFF
diff --git a/Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language.cpp b/Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language.cpp
index d909767526..3d235c1014 100644
--- a/Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language.cpp
+++ b/Marlin/src/lcd/extui/ftdi_eve_touch_ui/language/language.cpp
@@ -22,6 +22,8 @@
 
 #include "../../../../MarlinCore.h"
 
+#if ENABLED(TOUCH_UI_FTDI_EVE)
 #include "language.h"
 
 uint8_t lang = 0;
+#endif

```

ellensp, 2021-10-03:
As seriously consider upgrading to vscode and platfomio, its 10 times faster to compile compared with arduino ide.
It also automatically fetches all required librarayes.

closed 2021-10-04 by thinkyhead

dday58, 2021-10-31:
I upgrade to vscode and platformio yesterday and still get  error. So no vscode does not fix issue
.

thisiskeithb, 2021-10-31:
> I upgrade to vscode and platformio yesterday and still get error. So no vscode does not fix issue

This is not a Marlin issue.

<details><summary>Default config builds fine under the latest bugfix-2.0.x (9176dce):</summary>
<p>

```prolog
> Executing task: platformio run <

Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
----------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr 5.1.0 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio/build/mega2560/src/src/HAL/AVR/HAL.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/HAL_SPI.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/MarlinSerial.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/Servo.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/eeprom.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/fast_pwm.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/fastio.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/watchdog.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/Delay.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/HAL_MinSerial.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/eeprom_api.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/esp_wifi.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/shared/servo.cpp.o
Compiling .pio/build/mega2560/src/src/MarlinCore.cpp.o
Compiling .pio/build/mega2560/src/src/core/serial.cpp.o
Compiling .pio/build/mega2560/src/src/core/utility.cpp.o
Compiling .pio/build/mega2560/src/src/feature/ammeter.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/calibrate/G28.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M200-M205.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M220.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M221.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M301.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M302.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M309.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/config/M92.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M108_M112_M410.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M111.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M120_M121.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M17_M18_M84.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M211.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M400.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M80_M81.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M85.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M993_M994.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M997.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/M999.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/control/T.cpp.o
Compiling .pio/build/mega2560/src/src/gcode/eeprom/M500-M504.cpp.o
[... 91 lines cut ...]
Building .pio/build/mega2560/firmware.hex
================================ [SUCCESS] Took 17.80 seconds ================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       SUCCESS   00:00:17.802
================================ 1 succeeded in 00:00:17.802 ================================

Terminal will be reused by tasks, press any key to close it.
```

</p>
</details>

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).