MarlinFirmware/Marlin #22883: Delta auto calibration

renzaaa, 2021-10-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Delta auto calibration does not work. Probing 3 points and stop.
Older version of marlin was working.

### Version of Marlin Firmware

Latest bugfix-2.0.x or stable 2.0.x

### Printer model

Anycubic linear plus

### Electronics

skr 1.4 turbo

closed 2021-10-04 by renzaaa

maevebaksa, 2021-10-10:
Hi @renzaaa I have this exact problem! How did you fix it as I see you closed it. Please help!

thisiskeithb, 2021-10-10:
> Hi @renzaaa I have this exact problem! How did you fix it as I see you closed it. Please help!

@maevebaksa Please see https://github.com/MarlinFirmware/Marlin/issues/22886

maevebaksa, 2021-10-10:
@thisiskeithb Thank you for this. I just found this thread maybe 1-2 minutes ago. Hopefully the proposed changes will be merged soon. Cheers!