MarlinFirmware/Marlin #22886: [BUG] Delta Anycubic Kossel Linear Plus - Delta Calibration broken in 2.0.9.2

ChinChillaSPb, 2021-10-05:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Using default branch 2.0.x and correct config from corect configuration branch (2.0.9.2) (in configuration.h/_adv.h it marked as: "CONFIGURATION_H_VERSION 02000902", compiled w/o errors in arduino ide and vs code ide w platformio. also w/o errors. The problem is when you try to delta cal it stops after 4 point of 4 probes, and says "stow z-probe". always. nothing helps.

i'll try "bugfix branch" now

i using Anycubic Kossel Linear Plus with Ultrabase hb and probe v2.

also init eeprom of course, restored def.sett.

also cold reboot.

### Bug Timeline

new old bug

### Steps to Reproduce

1. compile with correct conf by vs code or arduino ide (vs code + platformio - much faster)
2. init eeprom
3. try to delta calibration on Kossel Linear Plus (all other my 3dp is standart geometry)
4. it stops with "stow z-probe" error after 4 point of 4 probes.

### Version of Marlin Firmware

2.0.9.2

### Printer model

Anycubic Kossel Linear Plus

### Electronics

stock

### Add-ons

probe v2, ultrabase hb

### Bed Leveling

UBL Bilinear mesh

### Additional information & file uploads

stock example config from the 2.0.9.2 branch

viper93458, 2021-10-05:
Not exactly the same issue for me but sensorless PROBING with skr 1.1 pro and 2209s on a delta doesn't work. Plows into the bed. May be related may not be. Just throwing it out there  :(

Sensorless homing does work though.

ChinChillaSPb, 2021-10-05:
also now back to 2.0.9.1 and confirm that problem is gone.
so the problem is on the 2.0.9.2

ChinChillaSPb, 2021-10-05:
> Not exactly the same issue for me but sensorless PROBING with skr 1.1 pro and 2209s on a delta doesn't work. Plows into the bed. May be related may not be. Just throwing it out there :(
> 
> Sensorless homing does work though.

i see that the problem is not related to probe type (i think because probing uses the same gcode command).
as you can see, i'm also must back to 9.1 so i can recommend you to get back to 9.1 too for a while.

renzaaa, 2021-10-05:
The same problem for my anycubic delta printer with 2.0.9.2 marlin version.
Delta calibration in Tevo little monster stops on iteration 1.

viper93458, 2021-10-05:
I am hoping I'm not the only one with sensorless PROBING issues on the delta. :)

I will watch this thread for fun.

LVD-AC, 2021-10-05:
> also now back to 2.0.9.1 and confirm that problem is gone. so the problem is on the 2.0.9.2

Could you test PR #22795, we're working on it...

LVD-AC, 2021-10-05:
@ChinChillaSPb , @renzaaa 

> also now back to 2.0.9.1 and confirm that problem is gone. so the problem is on the 2.0.9.2

Could you test PR #22795, we're working on it...

LVD-AC, 2021-10-05:
@viper93458 

> Not exactly the same issue .... Plows into the bed.

My printer uses the nozzle and a microswitch as probe and plows into the bed as well since 2.0.8. See #22853 for trying to find the needle in the haystack.

renzaaa, 2021-10-05:
> @ChinChillaSPb , @renzaaa
> 
> > also now back to 2.0.9.1 and confirm that problem is gone. so the problem is on the 2.0.9.2
> 
> This is a known bug. Use PR #22795 to fix this.

Working now, thanks.

nionio6915, 2021-10-06:
Could you please remove you comment from my issue thread? Your probelm is not applicable to my submission. Thanks

LVD-AC, 2021-10-06:
@renzaaa I don't think this was intended for you; but glad to know the fix works for you.

sjasonsmith, 2021-11-25:
Closing, since this has been fixed in #22795.

closed 2021-11-25 by sjasonsmith