MarlinFirmware/Marlin #22887: [BUG] Impossible to define SOL1_PIN in Marlin version 2.0.9.2

NuClear235, 2021-10-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Impossible to define SOL1_PIN in Marlin version 2.0.9.2
It was reported already on this thread, but not solved yet.
https://github.com/MarlinFirmware/Configurations/issues/581

### Version of Marlin Firmware

2.0.9.2

### Printer model

sCube

### Your Slicer

Simplify3D

### Host Software

Repetier Host

DerAndere1, 2021-10-06:
I had a probably related issue. I could not enable SOLENOID_PROBE, so as a workaround, I had to add the following, e.g. in Configuration.h 
`#define HAS_SOLENOID_1 1`

A slightly better fix, would be 
https://github.com/MarlinFirmware/Marlin/blob/9922410a73fd3285bfba0478ac16f04eeee48b8e/Marlin/src/inc/Conditionals_post.h#L2910
change that to 
```
#if ENABLED(SOLENOID_PROBE)
  #define HAS_SOLENOID_1 1
```
o

sjasonsmith, 2021-11-25:
I have verified that I can compile the configurations from the referenced issue. Since there have been no more replies I am going to close this, presuming this was fixed by the earlier mentioned commit.

closed 2021-11-25 by sjasonsmith