MarlinFirmware/Marlin #22888: [BUG] Printer Freezes when powerloss recovery is enabled

Sebastian-m-p, 2021-10-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Enabled powerloss recovery with marlin 2.0.9 and the printer just freezes randomly with the message Power outage please reset printer. tried with bugfix 2.0.x and the problem is still there print and freezes randomly. even when I press resume after resetting the message appear. 

need to reflash firmware because when the controller was set to off problem still appear

### Bug Timeline

Random

### Expected behavior

Expected to not do random power outtage and recover when real power outage happen

### Actual behavior

random power outage message where there is no power outtage at all

### Steps to Reproduce

enable power loss recovery 

### Version of Marlin Firmware

BugFix 2.0.x October 3rd 2021

### Printer model

Ender 3 Peo

### Electronics

SKR mini E3 V2

### Add-ons

Just added Neopixel (but the problem stars before it)

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/7284396/config.zip)

ellensp, 2021-10-05:
Your motherboard has a dedicated POWER_LOSS_PIN, so the instant you enable POWER_LOSS_RECOVERY ist starts looking at that pin.
You need to update your Configuration_adv.h to disable that pin.
Ie
#define POWER_LOSS_PIN         -1

closed 2021-10-05 by ellensp

Sebastian-m-p, 2021-10-05:
> Your motherboard has a dedicated POWER_LOSS_PIN, so the instant you enable POWER_LOSS_RECOVERY it starts looking at that pin. You need to update your Configuration_adv.h to disable that pin. Ie #define POWER_LOSS_PIN -1

i never even define the powe loss pin is that effecting it?

ellensp, 2021-10-05:
It is in Marlin/src/pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h as your motherboard has this pin.
So it is always defined the instant you selectyour motherboard.
Unless you explicit;y disable it by setting the pin to -1

Sebastian-m-p, 2021-10-05:
alright ca we please moce to chat on the marlin discrod? to not make this thread long