MarlinFirmware/Marlin #22893: [BUG] 2.0.9.2 PID

rq3, 2021-10-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

2.0.9.2 will not settle on a hot end temp. PID settings does not fix the issue.

### Bug Timeline

new with release 2.0.9.2. OK with 2.0.9.0

### Expected behavior

Temps should settle

### Actual behavior

Temps wander rapidly, and will not settle at assigned value. Print will not start.

### Steps to Reproduce

Compile and install Marlin 2.0.9.2 using released version and examples.

### Version of Marlin Firmware

2.0.9.2

### Printer model

Anycubic Predator

### Electronics

stock Trigorilla Pro

KayIO, 2021-10-29:
+1
On Trigorilla Pro board. the temperatures are instantly jumping +/- 5 ...therefore the print will not start due to missing settled temprature.
Works fine up to 2.0.9.1 ... but not in 2.0.9.2.

Alexgrosy, 2021-11-08:
+1
Same issue here with Trigorilla Pro. Temperature reading of hotend is totaly going nuts in 2.0.9.2.
2.0.9.1 is good.

ellensp, 2021-11-08:
If you enable SHOW_TEMP_ADC_VALUES and watch temperature in a console, does the raw ADC values also jump around?

Alexgrosy, 2021-11-08:
Just checked with latest bugfix code. The raw ADC values are equally jumpy.
https://pastebin.com/ECveATw7

rq3, 2022-01-15:
Problem still exists as of January 14, 2022 bugfix. Applies to both user interfaces ( classic and color ui), and is particularly bad with any RTD (platinum resistance sensors) where temps and ADC readings jump as much as +/- 15C between readings.

Appears to be an ADC issue with low impedance sensors.

KayIO, 2022-01-15:
> Problem still exists as of January 14, 2022 bugfix.

Confirmed ! I'm wondering why it worked before, but not now anymore

Michi1901, 2022-01-15:
Same here. My PT1000 is reading anywhere within +/- 5C of the target temperature. I am on 2.0.9.2.

Arakon, 2022-01-18:
While I'm not seeing jumps that badly, I've been having huge issues with my SKR E3 Mini 2.0 on 2.9.0.3. The temperature is jumping +/- 2° with sometimes unexplained sudden drops by up to 8°, causing a thermal runaway error. 4 out of 5 PID tune runs result in completely bizarre values that cause huge oscillations, too.
The bed is also unstable, but by far not that badly (+/- 0.8°).

2.9.0.3:
[image]

I just flashed the BTT 2.8.0.2 Marlin version for an unrelated issue, and all of a sudden, after a single PID tuning run, both bed and hotend are absolutely smooth, bed staying within 0.1°C of the set temperature, hotend within 1°, usually less than 0.5° off. 

2.8.0.2:
[image]

Thinkersbluff, 2022-02-05:
FWIW - I had a problem that sounds like this one, on my CR6-SE, when I flashed a "bugfix"-like developmental version of the CR6Community extui branch, which was recently merged with Marlin 2.9.0.1.   I could not get Load Filament to work, and teh nozzle temperature was jumping around the target value, seemingly sometimes heating or cooling faster than should be possible. 
Reverting to the released v6.1 of that firmware solved the problem.
I have a BTT SKR CR6 v1 board, a PT1000 thermistor and a Dragon HF hotend.

ceinem, 2022-02-23:
I'm observing a similar issue on the Anet A8+ with the latest Marlin firmware release.

descipher, 2022-02-23:
> FWIW - I had a problem that sounds like this one, on my CR6-SE, when I flashed a "bugfix"-like developmental version of the CR6Community extui branch, which was recently merged with Marlin 2.9.0.1. I could not get Load Filament to work, and the nozzle temperature was jumping around the target value, seemingly sometimes heating or cooling faster than should be possible. Reverting to the released v6.1 of that firmware solved the problem. I have a BTT SKR CR6 v1 board, a PT1000 thermistor and a Dragon HF hotend.

Does adding the following to configuration.h change the behavior?
#define ADC_RESOLUTION 10

2.0.9.1 -> 2.0.9.2 ADC resolution changed to 12 bit from 10 bits

Thinkersbluff, 2022-03-01:
> Does adding the following to configuration.h change the behavior?
> #define ADC_RESOLUTION 10

Sorry for the delayed response.  Deeply into a project the past 2 weeks...

Unfortunately, no, adding that line did not solve the problem.  
I still could not get Load/Unload to work, with the temperature fluctuating constantly above & below the target of 200C, with the PT1000 thermistor and the BTT CR6 SKR v1.0 motherboard.

It would not surprise me if the BTT SKR board itself has ADC issues if there is noise on the signal ground.  They do seem to have that issue on other SKR boards,  and I am not about to start de-soldering and replacing  SMD components on a board I can no longer replace, if I don't have to.

My problem does seem to be solved, though, without adding the line "#define ADC_RESOLUTION 10" to my Configuration.h, now that I have removed the PT1000 thermistor and I have re-installed the PT100.

InsanityAutomation, 2022-03-01:
On the couple machines I have with temp fluctuations like this where it was swinging a little due to the sensor resolution, I opened up #define TEMP_WINDOW to around 5c as the swing was 3c

tombrazier, 2022-03-03:
I would be very interested to know how my model predictive control algorithm handles such noisy sensor readings. It is an alternative to PID which should do better. PR #23751.

rq3, 2022-03-04:
> I would be very interested to know how my model predictive control algorithm handles such noisy sensor readings. It is an alternative to PID which should do better. PR #23751.
> 
> Not that PID is at fault for bad input. But I'd still be interested.

It compiled and installed fine, and does work, but the nozzle temp seems to wander a bit more. Oddly, in both cases (your algorithm and PID), when starting a print the printer seems to stick in an infinite settling loop, waiting for the nozzle temp to stabilize. If I then change the requested nozzle temp by just one degree, either up or down, the print starts instantly.

tombrazier, 2022-03-04:
> If I then change the requested nozzle temp by just one degree, either up or down, the print starts instantly.

I guess this behavior is an accidental byproduct of changing the target temp. Handy though!

If you're up for more experimentation...

Did you measure and set your own values for `AMBIENT_FOR_CALIBRATION`, `TEMPERATURE_AT_T10`, `TEMPERATURE_AT_T20` and `PWM_AT_200C`? What values did you get? It sounds like `sensor_xfer_coeff`, which is calculated from them, might be too large.

Another thought is to try changing the value 10.0 in line 1227 of [`Marlin/src/module/temperature.cpp`](https://github.com/MarlinFirmware/Marlin/pull/23751/files#diff-9c1d09e39e0d06d83cd4a44f54ff41e1cc364477c4dc39ecacf106a9b75540b7). Higher values will give greater immunity to sensor noise. However I think the problem is more likely to be with `sensor_xfer_coeff`.

rq3, 2022-03-04:
> I guess this behavior is an accidental byproduct of changing the target temp. Handy though!
>
> If you're up for more experimentation...
>
> Did you measure and set your own values for [...]

I did, and I'm always up for experimentation. Heater is 40 watts, 24 volts. Ambient temp is 19C. T10 was 46C, T20 was 88C. Using Repetier Host, my best guess for PWM was 125 (about 50%). Temperature plot at 200C oscillated slowly and repeatably +/- 3C.

tombrazier, 2022-03-04:
Wow, that's fast heating. I take it you started at T0 with ambient temperature?

The value of 125 will be wrong. Max PWM is 127. On my 36W E3D V6 clone, `PWM_AT_200C` is 45. `M105` will give you the value (after the `@` at the endish of the line). e.g. I get
```
ok T:21.05 /0.00 B:20.10 /0.00 P:19.53 /0.00 @:0 B@:0
```
The first `@` is the one - in this case 0 because I was not heating. The `B@` is for the bed.

Still, I would expect that to give a fixed offset, not oscillations. Will ponder and play a bit. In addition to getting a better PWM value, you might want to try fiddling with that 10.0 I mentioned above.

In the meantime, it occurred to me that you could narrow down the commit between 2.0.9.0 and 2.0.9.2 where your PID behaviour changed by doing a binary search through the revision tree. The revision in Configuration.h changed to 02000901 in commit 7726af9c534. So start about halfway between then and when you first noticed the change and see whether it is present, rinse repeat until you have the exact commit.

tombrazier, 2022-03-04:
Hang on, just a thought, MPCTEMP is disabled by default in my PR. You did enabled it right?

rq3, 2022-03-04:
Yes, MPCTEMP was enabled. I didn't know that max PWM was 127, so assuming it was 255, I used 125 as a roughly 50% power level. That probably explains why the temperature settled about 10C too high!

After dropping to your default PWM of 56, things went better, and the printer started well, but then the nozzle temperature slowly sagged from the assigned temperature of 195. I stopped the print after several minutes when the temp got down to 180.

For what it's worth, the Repetier temp plot is much more "ragged" looking than PID, yet the printer LCD display readout appears much more stable. Which may explain why the print actually started without having to force a start by monkeying with the nozzle temp.

tombrazier, 2022-03-05:
Getting back to the original issue, it looks to me like the problem is more about the measurement of temperature than PID. Nailing down the commit in the history of bugfix-2.0.x where the behaviour changed would be helpful.

Arakon, 2022-03-05:
For me, it actually turned out to be a failing power supply. I'm not sure why it was better with 2.0.8.2 vs. 2.0.9.3, but I eventually found that my PSU was dropping to 19V after a few mins under load and that caused the issues.

tombrazier, 2022-03-05:
> it actually turned out to be a failing power supply. I'm not sure why it was better with 2.0.8.2 vs. 2.0.9.3

Hmm, I wonder whether some firmware change increased hardware noise. A bit edge case but when you consider the 3D printers have long wires leading to sensors running in parallel with long wires leading to heaters and steppers, possible. @rq3 do you have an oscilloscope? Could you measure the actual electrical signal and see if there is a difference between versions of Marlin?

rq3, 2022-03-05:
> Hmm, I wonder whether some firmware change increased hardware noise. A bit edge case but when you consider the 3D printers have long wires leading to sensors running in parallel with long wires [...]

I do have a 500MHz scope, and used it extensively during development of the PT1000 amplifier. The amplifier feeds the ADC on the control board, and has a typical noise of about 20 microvolts.

Marlin started having ADC issues around release 2.0.9.2, when the ADC routines went from 10 bit to the correct 12 bit.

tombrazier, 2022-03-05:
Commit 5d8ca7c9445dac3d8bb52eafd9c45826e9c3387b removed a 10 bit bitmask. What happens if you insert a 12 bit one? i.e.
```
HAL_adc_result = (HAL_adc_results[(int)pin_index] >> (12 - HAL_ADC_RESOLUTION)) & 0xFFF; // shift out unused bits
```

rq3, 2022-03-06:
> Commit [5d8ca7c](https://github.com/MarlinFirmware/Marlin/commit/5d8ca7c9445dac3d8bb52eafd9c45826e9c3387b) removed a 10 bit bitmask. What happens if you insert a 12 bit one? i.e.
> 
> ```
> HAL_adc_result = (HAL_adc_results[(int)pin_index] >> (12 - HAL_ADC_RESOLUTION)) & 0xFFF; // shift out unused bits
> ```

While Marlin's PID issues have not been resolved, I find that the use of MPC as pushed by @tombrazier solves all of my issues.
I recommend that it be Marlin's default hotend control algorithm going forward, at least until Marlin gets a handle on its current PID implimentation.

tombrazier, 2022-03-06:
> I find that the use of MPC ... solves all of my issues.

Glad to hear that MPC handles the spurious sensor readings much better than PID. If anyone else wants to try it, I am keen for more feedback. The latest code for the MPC the pull request is [here](https://github.com/tombrazier/Marlin/archive/b70ba7a0507618dce31626552c747b44cdbe7341.zip).

However the original ADC issue is still on my mind. There are more things we can explore...

@rq3 has suggested to me that it may be related to how his mainboard cycles all the inputs through one ADC channel - perhaps not giving time for them to settle. I could believe the change to 12 bit might affect this. For a start, I guess sampling 12 bits takes longer than 10 and this may affect settling time in between. Then there is this in `Marlin/src/module/thermistor/thermistors.h:`
```
#if ENABLED(HAL_ADC_FILTERED)
  #define OVERSAMPLENR 1
#elif HAL_ADC_RESOLUTION > 10
  #define OVERSAMPLENR (20 - HAL_ADC_RESOLUTION)
#else
  #define OVERSAMPLENR 16
#endif
```

i.e. for 12 bit, only 8 samples are taken and averaged, whereas for 10 bit is was 16.

It would be really great to narrow down the actual offending commit. Anyone fancy trying f46e05fcb1e8b9a4ed36d2caa09122e75188e9da to see if it has the same problem? Code [here](https://github.com/MarlinFirmware/Marlin/archive/f46e05fcb1e8b9a4ed36d2caa09122e75188e9da.zip).

Incidentally, multiplexing onto a single ADC channel is how the AVR microprocessors work so this issue may have wider applicability than just particular 32 bit boards.

descipher, 2022-03-06:
ADC sampling at 12 vs 10 bits on the STM32F103 takes about 20ns longer per sample.

rq3, 2022-03-06:
> > I find that the use of MPC ... solves all of my issues.
> 
> Glad to hear that MPC handles the spurious sensor readings much better than PID. If anyone else wants to try it, I am keen for more feedback. The latest code for the MPC the pull request is [here](https://github.com/tombrazier/Marlin/archive/b70ba7a0507618dce31626552c747b44cdbe7341.zip).
> 
> However the original ADC issue is still on my mind. There are more things we can explore...
> 
> @rq3 has suggested to me that it may be related to how his mainboard cycles all the inputs through one ADC channel - perhaps not giving time for them to settle. I could believe the change to 12 bit might affect this. For a start, I guess sampling 12 bits takes longer than 10 and this may affect settling time in between. Then there is this in `Marlin/src/module/thermistor/thermistors.h:`
> 
> ```
> #if ENABLED(HAL_ADC_FILTERED)
>   #define OVERSAMPLENR 1
> #elif HAL_ADC_RESOLUTION > 10
>   #define OVERSAMPLENR (20 - HAL_ADC_RESOLUTION)
> #else
>   #define OVERSAMPLENR 16
> #endif
> ```
> 
> i.e. for 12 bit, only 8 samples are taken and averaged, whereas for 10 bit is was 16.
> 
> It would be really great to narrow down the actual offending commit. Anyone fancy trying [f46e05f](https://github.com/MarlinFirmware/Marlin/commit/f46e05fcb1e8b9a4ed36d2caa09122e75188e9da) to see if it has the same problem? Code [here](https://github.com/MarlinFirmware/Marlin/archive/f46e05fcb1e8b9a4ed36d2caa09122e75188e9da.zip).
> 
> Incidentally, multiplexing onto a single ADC channel is how the AVR microprocessors work so this issue may have wider applicability than just particular 32 bit boards.

I don't know how many times I've looked at thermistors.h, but thanks to your kick to my head, I brute forced the oversampling to 16 (for my 12 bit ADC) by changing
#define OVERSAMPLENR (20 - HAL_ADC_RESOLUTION)
to
#define OVERSAMPLENR (28 - HAL_ADC_RESOLUTION)

All of the plotted ADC bouncing is effectively gone, and the temperature settles rapidly to within hundredths of a degree.
I even went overboard and tried
#define OVERSAMPLENR (36 - HAL_ADC_RESOLUTION) to force 24x oversampling, but saw no improvement.

Thank you. I am not aware of any 32 bit processors using ADC's that are NOT 12 bit, so I believe this could be a permanent fix within Marlin for the on-going ADC issues.

tombrazier, 2022-03-06:
Right. That code is explained [here](https://github.com/MarlinFirmware/Marlin/pull/20519#pullrequestreview-556221282). I'll have a look at what might be done.

descipher, 2022-03-06:
Makes no sense to me, with 12 bits there is no overflow at 16 samples.
Maximum 4095 x 16 into uint16_t is ok. Only More than 12 bits need to be handled.
28 - HAL_ADC_RESOLUTION would be the better option.

rq3, 2022-03-06:
> Makes no sense to me, with 12 bits there is no overflow at 16 samples. Maximum 4095 x 16 into uint16_t is ok. Only More than 12 bits need to be handled. 28 - HAL_ADC_RESOLUTION would be the better option.

Read my post again. I am using 28 - HAL_ADC_RESOLUTION.
I have also found that modifying src/HAL/STM32F1/HAL.cpp from:
HAL_adc_result = HAL_adc_results[(int)pin_index] >> (12 - HAL_ADC_RESOLUTION); // shift out unused bits
to:
HAL_adc_result = HAL_adc_results[(int)pin_index] >> (28 - HAL_ADC_RESOLUTION); // shift out unused bits
appears to provide even more stability and smoothing.

I believe that these two changes, along with the MPC algorithm by @tombrazier, puts all of my issues to bed

rq3, 2022-03-06:
> Right. That code is explained [here](https://github.com/MarlinFirmware/Marlin/pull/20519#pullrequestreview-556221282). I'll have a look at what might be done.

Yes, I've followed that review with interest. My gut feeling is that there was a stumble between the days when almost all ADC's were 10 bit, and today when 12 bit (or more) is the norm.

descipher, 2022-03-06:
I am referring to the logic in the PR that created 20 - HAL_ADC_RESOLUTION. It was wrong to do that.

tombrazier, 2022-03-06:
> I have also found that modifying src/HAL/STM32F1/HAL.cpp from:
HAL_adc_result = HAL_adc_results[(int)pin_index] >> (12 - HAL_ADC_RESOLUTION); // shift out unused bits
to:
HAL_adc_result = HAL_adc_results[(int)pin_index] >> (28 - HAL_ADC_RESOLUTION); // shift out unused bits
appears to provide even more stability and smoothing.

@rq3 Warning: I think you might just be throwing away your entire ADC result and getting zero regardless of the temperature. If I'm right there is a fire risk. No time to look more closely at the code now to see whether I am right so I might be mistaken. But best to be on the safe side and undo that change.

tombrazier, 2022-03-06:
> Makes no sense to me, with 12 bits there is no overflow at 16 samples.
Maximum 4095 x 16 into uint16_t is ok. Only More than 12 bits need to be handled.
28 - HAL_ADC_RESOLUTION would be the better option.

@descipher Agreed it makes no sense. But the right code is
```
#elif HAL_ADC_RESOLUTION > 12
  #define OVERSAMPLENR (1 << (16 - HAL_ADC_RESOLUTION))
```

I think. Will get onto a PR tomorrow.

descipher, 2022-03-06:
#elif HAL_ADC_RESOLUTION > 12
  #define OVERSAMPLENR (1 << (16 - HAL_ADC_RESOLUTION))

I agree, that should work.  Left shift 4 into OVERSAMPLENR = 16 samples or less with more resolution.

descipher, 2022-03-06:
I will test this on the STM32F103.

```
#elif HAL_ADC_RESOLUTION > 12
  #define OVERSAMPLENR (1 << (16 - HAL_ADC_RESOLUTION))
#else
  #define OVERSAMPLENR 16
#endif
```

tombrazier, 2022-03-07:
@descipher You didn't get a compile error?
```
// Currently Marlin stores all oversampled ADC values as int16_t, make sure the HAL settings do not overflow 15bit
#if MAX_RAW_THERMISTOR_VALUE > ((1 << 15) - 1)
  #error "MAX_RAW_THERMISTOR_VALUE is too large for int16_t. Reduce OVERSAMPLENR or HAL_ADC_RESOLUTION."
#endif
```

tombrazier, 2022-03-07:
Update on whether it makes sense to limit to 15 bits: `temperature.h` has
```
typedef struct TempInfo {
  uint16_t acc;
  int16_t raw;
  ...
  inline void update() { raw = acc; }
} temp_info_t;
```
So the constraint is on the variable `raw`. Why that is signed I do not yet know.

descipher, 2022-03-07:
> @descipher You didn't get a compile error?
>
> ```
> // Currently Marlin stores all oversampled ADC values as int16_t, make sure the HAL settings do not overflow 15bit
> #if MAX_RAW_THERMISTOR_VALUE > ((1 [...]

Using thermistor of type 1 did not trigger that condition.

tombrazier, 2022-03-07:
Ah. Perhaps an overflow hid the error. What if you use
```
#define MAX_RAW_THERMISTOR_VALUE ((uint32_t)(HAL_ADC_RANGE) * (OVERSAMPLENR) - 1)
```

descipher, 2022-03-07:
I think it would be better to use int32_t and uint32_t on 32 bit processors since they are 12bit+ capable.
If we don't it we will have the 8x oversample problem crop up anyway and we have improved nothing.
The other method would be to shift 12+ bits into a 10 bit space to work around the issue for now.

tombrazier, 2022-03-07:
I'm guessing 32 bit machines have plenty more RAM than the AVRs so no one would especially mind 32 bit variables.

I have had to change quite a lot of code to get the whole thing working with uint16_t. All the same code would need examining and changing for int32_t. Also, 16 bit really simplifies the ADC for filament width sensor and power supply current and voltage. These deliberately expand to 32 bit safe in the knowledge that their 16 bit input will not overflow the 32 bit variables.

I hope my PR is close to submitting. Then it can get a good kicking around.

descipher, 2022-03-07:
Just tested this, no issues on first run. Temps are stable.
```
// A temperature sensor
typedef struct TempInfo {
  TERN(CPU_32_BIT, uint32_t, uint16_t) acc;
  TERN(CPU_32_BIT, int32_t, int16_t) raw;
  celsius_float_t celsius;
  inline void reset() { acc = 0; }
  inline void sample(const TERN(CPU_32_BIT, uint32_t, uint16_t) s) { acc += s; }
  inline void update() { raw = acc; }
} temp_info_t;
```

descipher, 2022-03-07:
Oh, don't really need  ``` inline void sample(const TERN(CPU_32_BIT, uint32_t, uint16_t) s) { acc += s; }```

descipher, 2022-03-07:
Better:
```
// A temperature sensor
typedef struct TempInfo {
  TERN(CPU_32_BIT, uint32_t, uint16_t) acc;
  TERN(CPU_32_BIT, int32_t, int16_t) raw;
  celsius_float_t celsius;
  inline void reset() { acc = 0; }
  inline void sample(const uint16_t s) { acc += s; }
  inline void update() { raw = acc; }
} temp_info_t;
```

tombrazier, 2022-03-07:
@descipher Not that simple. All the uses of TempInfo::raw need checking. PR created for the uint16_t fix. Fancy running through all the changes changing uint16_t to your conditional uint32_t?

rq3, 2022-03-07:
> Using thermistor of type 1 did not trigger that condition.

Also worked fine with custom amplified PT1000.

descipher, 2022-03-07:
Yes, I know it's more that just defining the 32bit variables in the struct.
The overflow of sampling starts with ```sample(const uint16_t s) { acc += s; }``` in the struct. So it's the must have.

rq3, 2022-03-07:
> @rq3 Warning: I think you might just be throwing away your entire ADC result and getting zero regardless of the temperature. If I'm right there is a fire risk. No time to look more closely at the [...]

Thank you. I had not considered what might happen when the ADC data "rolls over" and, say, 500C becomes 0C. However, after setting the value back to the original 12, the ADC data looks almost bang-bang, with the massive (+/- 20 degree) random jumps. I have settled on 24, which should truncate at 12 bits with my 12 bit ADC (?), and Marlin seems to recognise fault conditions with an expected shutdown if I force an overtemp or disconnect the sensor.

One other quick thought regarding "future-proofing" Marlin. I have heard internal scuttlebutt from industrial sources, who shall remain un-named, that 22-24 bit ADC functionality will probably become "standard" fairly soon.

tombrazier, 2022-03-08:
@rq3 It is possible that your processor doesn't use the STM32F1 HAL code but rather the STM32. The `>>` operator is a bitshift and it divides by a power of 2. `HAL_adc_results[(int)pin_index] >> (24 - HAL_ADC_RESOLUTION)` will shift out 12 bits, dividing the 12 bit signal down to zero.

I should think extending ADC to 20+ bit would be really simple from a hardware perspective. But providing it with an analogue signal that has a sufficiently low signal to noise ratio is tricky. So it's not like we'll easily be able to measure, say, 200C to within a thousandth of a degree. But it could make it a lot simpler to interface with a very low impedance small signal like a thermocouple - you could just bung one straight on the ADC input without an amplifier.

Anyway, any bit depth above 12 will need the 32 bit extension @descipher has been experimenting with. I think it is probably as simple as

`typedef TERN(CPU_32_BIT, uint32_t, uint16_t) adc_acc_t;`

and then replace all the relevant `uint16_t`s with `adc_acc_t`s. But there also may be unexpected side effects.

descipher, 2022-03-08:
I use the STM32 HAL, maple is deprecated should not be used any more .. for those who may go there .. lol
HAL_ADC_RESOLUTION 12 is defined in various files such as the stm32g0.ini file so it still applies to a couple of boards.
```
# BigTree SKR mini E3 V3.0 (STM32G0B1RET6 ARM Cortex-M0+)
#
[env:STM32G0B1RE_btt]
extends                     = stm32_variant
platform                    = ststm32@~14.1.0
platform_packages           = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip
board                       = marlin_STM32G0B1RE
board_build.offset          = 0x2000
board_upload.offset_address = 0x08002000
build_flags                 = ${stm32_variant.build_flags}
                            -DADC_RESOLUTION=12
```
Like the SKR MINI .. which one user has indicated issues with. So there maybe a relational indication in that. 

Otherwise it would explain was there was absolutely no change when setting HAL_ADC_RESOLUTION 10 .. and no compile error.

tombrazier, 2022-03-08:
So the user with the SKR MINI may have had the issue with OVERSAMPLENR=8. I am now back to thinking the loss of the 10 bit bitmask might explain errors for some other users, even @rq3 who only tested with a 12 bit bitmask.

descipher, 2022-03-08:
Yes that is a possibility. Was the masking just cleaning outlier values or other inserted bad values?  Or is there a global variable overwrite/overflow problem it masking.

tombrazier, 2022-03-08:
@rq3 any idea whether you are building against STM32 or STM32F1? If you are using platformio, it will be the difference between `default_envs = trigorilla_pro` and `default_envs = trigorilla_pro_maple` in platformio,ini.

@descipher the memory the mask is applied to gets written by DMA. From what I can tell ADC runs in a continuous cycle. Who knows what is in the upper 4 bits? Quite possibly just uninitialised memory. (I think it is always 12 bit native and Marlin just shifts it down to 10 bits.)

rq3, 2022-03-08:
> @rq3 any idea whether you are building against STM32 or STM32F1? If you are using platformio, it will be the difference between `default_envs = trigorilla_pro` and `default_envs = [...]

I am building under trigorilla_pro. I can say that these changes, inclusive of MPC, have made the ADC and temperatures infinitely more stable. Additionally, the ADC crosstalk, which used to be VERY evident when the bed heater was enabled and disabled, has effectively disappeared. I do not know if this is just a visual artifact from the way Repetier Host generates its plots, but the effect is very apparent.

tombrazier, 2022-03-08:
Retract which statement? The one about 12 bits?

tombrazier, 2022-03-08:
Compiles and works with
```
#if ENABLED(HAL_ADC_FILTERED)
  #define OVERSAMPLENR 1
#elif HAL_ADC_RESOLUTION > 12
  #define OVERSAMPLENR (1 << (16 - HAL_ADC_RESOLUTION))
#else
  #define OVERSAMPLENR 16
#endif
```
in thermisters.h? I'm still expecting that #error statement to kick in.

rq3, 2022-03-08:
> Compiles and works with
> 
> ```
> #if ENABLED(HAL_ADC_FILTERED)
>   #define OVERSAMPLENR 1
> #elif HAL_ADC_RESOLUTION > 12
>   #define OVERSAMPLENR (1 << (16 - HAL_ADC_RESOLUTION))
> #else
>   #define OVERSAMPLENR 16
> #endif
> ```
> 
> in thermisters.h? I'm still expecting that #error statement to kick in.

No. Thermistors.h currently looks like:
#define THERMISTOR_TABLE_ADC_RESOLUTION 10
#define THERMISTOR_TABLE_SCALE (HAL_ADC_RANGE / _BV(THERMISTOR_TABLE_ADC_RESOLUTION))
#if ENABLED(HAL_ADC_FILTERED)
  #define OVERSAMPLENR 1
#else
  #define OVERSAMPLENR 16
#endif

Stand-by while I try the change.

[21 more replies]

closed 2022-03-12 by thisiskeithb