MarlinFirmware/Marlin #22895: Marlin Compilation Error On MKS ROBIN V2.4

Haiang, 2021-10-06:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Currently I am using MKS Robin V2.4 board and TFT V2.0 screen when I use this Marlin I compile I get error
Print the file included from .piolibdeps\mks_robin\TMCStepper\src\source\CHOPCONF.cpp:1:0:
.piolibdeps\mks_robin\TMCStepper\src/TMCStepper.h:38:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************** **********************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI > platformio lib search "header:SoftwareSerial.h"
* Web > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************** **********************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs\mks_robin\libe11\TMCStepper\source\DRVCONF.cpp.o
*** [.pioenvs\mks_robin\libe11\TMCStepper\source\CHOPCONF.cpp.o] Error 1
Print the file included from .piolibdeps\mks_robin\TMCStepper\src\source\COOLCONF.cpp:1:0:
.piolibdeps\mks_robin\TMCStepper\src/TMCStepper.h:38:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************** **********************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI > platformio lib search "header:SoftwareSerial.h"
* Web > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************** **********************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\mks_robin\libe11\TMCStepper\source\COOLCONF.cpp.o] Error 1
Print the file included from .piolibdeps\mks_robin\TMCStepper\src\source\DRVCONF.cpp:1:0:
.piolibdeps\mks_robin\TMCStepper\src/TMCStepper.h:38:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************** **********************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI > platformio lib search "header:SoftwareSerial.h"
* Web > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************** **********************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\mks_robin\libe11\TMCStepper\source\DRVCONF.cpp.o] Error 1
================================================== ======================================= [FAILED] Took 47.68 seconds ===== ================================================== ===============================================

Environment Status Duration
------------- -------- ------------
mks_robin FAILED 00:00:47.678
================================================== ===================================== 1 failed, 0 succeeded in 00:00:47.678 ===== ================================================== ==========================================
The terminal process "C:\Users\Administrator\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

### Expected behavior

Currently I use Visula Stuido Code and Platform IO to compile the code
In PlatformIO.ini I configure the following
If I set platrorm = stsm32, I get more errors

# MKS Robin (STM32F103ZET6)
#
[env:mks_robin]
platform = ststm32@<5.7.0
framework = arduino
board = genericSTM32F103ZE
extra_scripts = buildroot/share/PlatformIO/scripts/mks_robin.py
build_flags = !python Marlin/src/HAL/HAL_STM32F1/STM32F1_flag_script.py
   ${common.build_flags}
   -DSTM32_XL_DENSITY
src_filter = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
lib_deps = ${common.lib_deps}
lib_ignore = c1921b4
   libf3c
   lib066
   Adafruit NeoPixel_ID28
   Adafruit NeoPixel
   libf3e
   TMC26XStepper

### Steps to Reproduce

I hope someone can help me with this problem

### Version of Marlin Firmware

Marlin

### Electronics

MKS Robin V2.4

### Additional information & file uploads

[Marlin-MKS-Robin-Touch-POC.zip](https://github.com/MarlinFirmware/Marlin/files/7292637/Marlin-MKS-Robin-Touch-POC.zip)

ellensp, 2021-10-06:
Positively ancient version of marlin CONFIGURATION_H_VERSION 020000 bugfix-2.0.x

ellensp, 2021-10-06:
Seems to have come from https://github.com/jmz52/Marlin/tree/MKS-Robin-Touch-POC
This is a 3rd party fork of Marlin, with custom code modifications 
You need to get support from them.