MarlinFirmware/Marlin #22911: [BUG] (ModuleNotFoundError: No module named 'platformio.managers.package':)

Haiang, 2021-10-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello friends
Currently I am using Visual Studio Code version 1.61.0 and Plattrorm IO version v2.3.3 to program Marlin code but I am getting an error like the image I describe, even though I have tried to look at the posts and issues. topic of this error but I still can't compile it successfully. I use MKS Robin V2.4 board and Mks Robin TFT V2.0 board, hope you can help and fix this problem for me.
This is the path of the Marlin file I'm using
https://github.com/jmz52/Marlin/tree/R61502
Here is the link for the fix update
https://github.com/MarlinFirmware/Marlin/pull/18721/commits/cf9e6c9d797dd13020359f7db82242b8ebd96966
Thank you very much

### Bug Timeline

an old problem

### Expected behavior

I tried but it doesn't compile

### Version of Marlin Firmware

2.0.6

Haiang, 2021-10-09:
can you help me fix the error

ellensp, 2021-10-09:
No. It an obsolete version of marlin expecting an obsolete version of platformio.