MarlinFirmware/Marlin #22920: Idex travels into left endstop[BUG] (bug summary)

shtaXveed, 2021-10-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

following the update with the new bugfix, the heads still travel at the same time at the beginning of a print but seems to work fine while outside of print?

### Expected behavior

right hot end travels to start position while left if stationary
no grinding.

### Actual behavior

right travels to start location but left travels at same time into endstop.

### Steps to Reproduce

1 switch idex modes. 
2 watch it line up properly
3 start print and wait for warm up
4 start new print
5 printer rehomes with bl touch to home both hotends travel back to end stops then travel to stat position. while traveling left hits endstop 

### Version of Marlin Firmware

bugfix 2.0.9.2

### Printer model

jgmaker artist d pro

### Electronics

stock electronics plus bltouch

### Add-ons

bltouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

InsanityAutomation, 2021-10-10:
Can you attach both you're configuration files and a sample gcode file that exhibits the issue?

InsanityAutomation, 2021-10-10:
I just went through the GCode and I did not see M605 in the starting code to set the mode anywhere. I added the following after G28 : M605S2; G28X;

With that, it started up correctly and ran here. 
Im am able to reproduce setting the mode from the menu before calling this print, but I believe the G28 with no args while in duplicate mode is whats causing the issue.

Considering you do not have the forced mode apply in configuration_adv, and are not setting up the mode completely in the gcode, I can see why this would behave strangely. 

// Default action to execute following M605 mode change commands. Typically G28X to apply new mode.
  //#define EVENT_GCODE_IDEX_AFTER_MODECHANGE "G28X"

Workaround for the moment is to add the mode set commands to the gcode after the G28 as stated above. Ill look at whats wrong with the position sets persisting when mode is set prior to entering G28.

shtaXveed, 2021-10-11:
oh, thats my fault then, i was coping off a previous version and must have
missed that

InsanityAutomation, 2021-10-11:
From the menu, or set prior still should have worked so not a big deal, but the fix for that is up now too! Means it was essentially a second related bug.

InsanityAutomation, 2021-10-13:
> Does the extra benefit of `G28X` after `M605` need to be applied as part of the `EVENT_GCODE_IDEX_AFTER_MODECHANGE` option, or could the behavior that comes from adding `G28X` here be applied differently, perhaps as a standard thing for certain cases? Alternatively, if `EVENT_GCODE_IDEX_AFTER_MODECHANGE` is not enabled, or doesn't contain `G28X`, we could throw an additional warning to the user. Whatever the case may be, the "out of the box" behavior should be a working one.

It is necessary to set up the offset positions, and that's why when I originally put it in I made it that gcode and enabled by default. There was originally some apprehension about mode change calling the motion. I don't really have any aversion to adding a warning or just making it a built in action myself.

thinkyhead, 2026-10-08:
This looks to have been addressed by #22925, merged since this report, and there's been no further activity here. Closing for now as part of 2.1.3 cleanup. If you still see the problem on current `bugfix-2.1.x`, please let us know and we'll reopen.

closed 2026-10-08 by thinkyhead