MarlinFirmware/Marlin #22921: [BUG] Cold extrusion not working

ChristianNavolskyi, 2021-10-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I used the LCD screen to test all the axis on my Anycubic i3 Mega. After XYZ worked I wanted to test the extrusion stepper.
After confirming in the menu, that I want to proceed even though the hotend is not heated the motor is not moving at all when commanding it to move.

When disabling the `//#define PREVENT_COLD_EXTRUSION` flag it works as expected.

### Expected behavior

After confirming that I want to proceed even with the cold hotend, I expect the stepper motor to rotate when sending some movement commands to it.

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Oct 10 2021 17:15:20)

### Printer model

Anycubic i3 Mega (dual z-axis)

### Electronics

SKR v1.3, TMC2208 drivers

### Additional information & file uploads

The log says `echo: cold extrusion prevented` when trying to move the stepper motor.

**Configuration**
[cold extrusion.zip](https://github.com/MarlinFirmware/Marlin/files/7317719/bug.report.-.cold.extrusion.zip)

I set up the second z-axis for the E0 slot on the board and shifted the E0 config to the E1 pins on the board.

sjasonsmith, 2021-11-25:
This sounds like exactly what `PREVENT_COLD_EXTRUSION` is supposed to do.

I see that the override menu was added in the following pull request, but it isn't 100% clear to me whether it was actually effective. It looks like it would allow you to get into the movement menu, but I'm not sure that it would actually bypass the lower-level motion/planner checks. It is possible that it did when it was originally added, but has been broken in refactoring since then.
#19606

thisiskeithb, 2021-11-26:
> I see that the override menu was added in the following pull request, but it isn't 100% clear to me whether it was actually effective.

> It is possible that it did when it was originally added, but has been broken in refactoring since then.

It worked at one point, but I ran into this very issue while testing some new board/driver combos with the latest `bugfix-2.0.x` tonight.

ellensp, 2022-04-09:
I just reviewed the current code.

All instances that could print out  STR_ERR_COLD_EXTRUDE_STOP  ie  " cold extrusion prevented"
Are in       #if ENABLED(PREVENT_COLD_EXTRUSION) code blocks are are correctly disabled.

I also verified this on a BOARD_SANGUINOLOLU_12 (I had it already setup form previous issue I was looking at) 
Disabled #define PREVENT_COLD_EXTRUSION 
And E0 moves fine without issues.

Current temperature is 
M105 
> ok T:-15.00 /0.00 @:0

This seems to have already been fixed.

thisiskeithb, 2022-04-09:
I can still reproduce this on current bugfix with my BTT002/TMC2209 build.

When cold and `PREVENT_COLD_EXTRUSION` enabled (per defaults), I'm asked if I want to proceed while trying to move E when cold, but the E motor doesn't move unless up to temp.

ellensp, 2022-04-09:
Just tested on ramps as well both via gcode and LCD motion menu. Works on this system also. 
Exploring 32 bit systems now.

ellensp, 2022-04-09:
[Bugfix_Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/8457396/Bugfix_Configurations.zip)
Updated to bugfix Configs and pins files for testing

ellensp, 2022-04-10:
Using these updated  configs on a skr1.3   PREVENT_COLD_EXTRUSION works as expected.  (yes Im even using your pins file, so its E1 that is moving
Only change I made was using  tmc2209's as I don't have tmc2208's 

I cannot make this fail.

thisiskeithb, 2022-04-10:
> Using your configs (updated to current bugfix) on a real skr1.3 PREVENT_COLD_EXTRUSION works as expected.

The point is that you were still given the option to bypass cold extrusion and move E, even when `PREVENT_COLD_EXTRUSION` enabled, with that “proceed?” message when selecting move E while cold.

Otherwise, what is the point of giving users the option to bypass?

This feature worked fine at one point, even with `PREVENT_COLD_EXTRUSION` enabled.

ellensp, 2022-04-10:
After clearing up the confusion on my part.  I can confirm the issue. 

a partial solution.  Might inspire someone while I sleep,

```DIFF
--- a/Marlin/src/lcd/menu/menu_motion.cpp
+++ b/Marlin/src/lcd/menu/menu_motion.cpp
@@ -227,9 +227,11 @@ void _menu_move_distance(const AxisEnum axis, const screenFunc_t func, const int
       if (too_cold) {
         ui.goto_screen([]{
           MenuItem_confirm::select_screen(
-            GET_TEXT(MSG_BUTTON_PROCEED), GET_TEXT(MSG_BACK),
-            _goto_menu_move_distance_e, nullptr,
-            GET_TEXT(MSG_HOTEND_TOO_COLD), (const char *)nullptr, PSTR("!")
+            GET_TEXT(MSG_BUTTON_PROCEED), GET_TEXT(MSG_BACK)
+            , [] { Temperature::allow_cold_extrude = true; _goto_menu_move_distance_e(); }
+            , nullptr
+            , GET_TEXT(MSG_HOTEND_TOO_COLD)
+            , (const char *)nullptr, PSTR("!")
           );
         });
         return;
``` 
This works, but allow_cold_extrude remains set set This should reset when you exit the menu, I thin It should re-prompt you every time. you try and move E when cold.

ellensp, 2022-04-15:
This PR should fix it. Closing

thisiskeithb, 2022-05-15:
Reopening since https://github.com/MarlinFirmware/Marlin/pull/24045 was broken right before it was merged.

https://github.com/MarlinFirmware/Marlin/pull/24045#issuecomment-1126873301

ellensp, 2022-05-15:
Confirmed was broken again by the rogue programmer.

This seems to fix it.  Still testing, but looking good.

```DIFF
diff --git a/Marlin/src/lcd/menu/menu.cpp b/Marlin/src/lcd/menu/menu.cpp
index d56bc9b5ff..8157acdfa0 100644
--- a/Marlin/src/lcd/menu/menu.cpp
+++ b/Marlin/src/lcd/menu/menu.cpp
@@ -168,8 +168,6 @@ void MenuEditItemBase::goto_edit_screen(
  */
 void MarlinUI::goto_screen(screenFunc_t screen, const uint16_t encoder/*=0*/, const uint8_t top/*=0*/, const uint8_t items/*=0*/) {
   if (currentScreen != screen) {
-    thermalManager.set_menu_cold_override(false);
-
     TERN_(IS_DWIN_MARLINUI, did_first_redraw = false);
 
     TERN_(HAS_TOUCH_BUTTONS, repeat_delay = BUTTON_DELAY_MENU);
diff --git a/Marlin/src/lcd/menu/menu_motion.cpp b/Marlin/src/lcd/menu/menu_motion.cpp
index 1ef38c014b..765172b827 100644
--- a/Marlin/src/lcd/menu/menu_motion.cpp
+++ b/Marlin/src/lcd/menu/menu_motion.cpp
@@ -228,7 +228,7 @@ void _menu_move_distance(const AxisEnum axis, const screenFunc_t func, const int
         ui.goto_screen([]{
           MenuItem_confirm::select_screen(
             GET_TEXT(MSG_BUTTON_PROCEED), GET_TEXT(MSG_BACK),
-            [] { _goto_menu_move_distance_e(); thermalManager.set_menu_cold_override(true); }, nullptr,
+            [] { thermalManager.set_menu_cold_override(true); _goto_menu_move_distance_e(); }, nullptr,
             GET_TEXT(MSG_HOTEND_TOO_COLD), (const char *)nullptr, PSTR("!")
           );
         });
@@ -323,6 +323,8 @@ void menu_move() {
 
   #if E_MANUAL
 
+    thermalManager.set_menu_cold_override(false);
+
     // The current extruder
     SUBMENU(MSG_MOVE_E, []{ _menu_move_distance_e_maybe(); });
 
```

closed 2022-05-17 by thisiskeithb