MarlinFirmware/Marlin #22936: [BUG] MACROS SERIAL_ECHOPAIR and SERIAL_ECHOLNPAIR are missing in 2.0.9.2 (still present in 2.0.9.1)

tmartin007, 2021-10-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The preprocessor macros:

`#define SERIAL_ECHOPAIR(V...)     do{ EVAL(_SEP_N(TWO_ARGS(V),V)); }while(0)`

and 

`#define SERIAL_ECHOLNPAIR(V...)   do{ EVAL(_SELP_N(TWO_ARGS(V),V)); }while(0)`
 
are missing from 2.0.9.2 (breaking 2.0.9.1 code). 

### Expected behavior

The preprocessor macros:

`#define SERIAL_ECHOPAIR(V...)     do{ EVAL(_SEP_N(TWO_ARGS(V),V)); }while(0)`

and 

`#define SERIAL_ECHOLNPAIR(V...)   do{ EVAL(_SELP_N(TWO_ARGS(V),V)); }while(0)`
 
should be present in 2.0.9.2.

Alternatively, the release notes should document this and offer alternative.
A #pragma could provide a hint, too. 

### Actual behavior

compilation fails, if either of the macros is used.

### Steps to Reproduce

1. add `SERIAL_ECHOLNPAIR("Test:", 123);` to e.g. gcode.cpp line 325
2. compile marlin
3. observer the compiler's error message 

### Version of Marlin Firmware

2.0.9.2 

### Printer model

any

### Electronics

any

The-EG, 2021-10-11:
Looks like those got renamed to `DEBUG_ECHOPGM` and `DEBUG_ECHOLNPGM` respectively:

https://github.com/MarlinFirmware/Marlin/commit/b661795ae5af15d773b9c148abe1c8005799dac4#diff-a2827095111fe5cc7db08a45d86b8454406048e067f13a566088e7b7a2edd003L40-R40

tmartin007, 2021-10-11:
Sorry, I had misinterpreted the "_Fewer serial macros_" release note.