MarlinFirmware/Marlin #22963: [BUG] G34 Auto Align Z not working correctly (can't achieve accuracy)

bmyonatan, 2021-10-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When performing G34, the difference between Z2 and Z1 is very unreliable and doesn't converge to 0 (sometimes accuracy increases between iterations and sometimes it decreases).
I have an Ender 3 Pro, with a SKR 2 board, 2209 drivers, dual Z lead screw and drivers and a BLTouch.
I tried different number of iterations and amplitudes, but the auto align is unreliable (rarely it will get to the desired accuracy, but usually Z2-Z1 is above 1.5

Bltouch accuracy (with M48 S1) is <0.003
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7357338/configuration.zip)

### Expected behavior

Both Z motors align themselves properly

### Actual behavior

Z alignment is not working

### Steps to Reproduce

Running G34 on my printer with supplied config

### Version of Marlin Firmware

Bugfix-2.0.x (2021-10-16_

### Printer model

Ender 3 Pro

### Electronics

SKR 2.0 with 2209's

### Add-ons

BLTouch, dual z lead screws (+motors and drivers)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

gknapp, 2021-11-04:
FWIW, I'm also running an SKR E3 Turbo and using G34 am able to achieve an accuracy of 0.02. I had to invert the second (newer) stepper motion I got from an ali express kit.

ashinckley, 2022-01-18:
> > I am experiencing the same behavior. Ender 3 with SKR E3 turbo. I tried switching cables and still, but Z2-Z1 is above 1.6
> 
> How do yo have Z2 connected? Is it hooked up to the second Z port or to the E1 port? If you connect to the second Z port both motors are running off the same driver and `G34` won't work at all.

Thank you so much for this.  That should have been the first thing I checked, but for some reason I went through a deep-dive into my marlin configuration instead.

closed 2022-04-05 by github-actions[bot]