MarlinFirmware/Marlin #22969: [BUG] BABYSTEP_ZPROBE_GFX_OVERLAY layout fail with DWIN_MARLINUI_LANDSCAPE

aalku, 2021-10-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

BABYSTEP_ZPROBE_GFX_OVERLAY layout fails with DWIN_MARLINUI_LANDSCAPE
[image]

### Bug Timeline

I think DWIN_MARLINUI_LANDSCAPE is a new feature so bug is new.

### Expected behavior

The icons should be better placed arround the screen.

### Actual behavior

The icons are overlapping with the text and the "progress bar" or whatever that rectangle is.

### Steps to Reproduce

1. Enable DWIN_MARLINUI_LANDSCAPE.
2. Enable BABYSTEP_ZPROBE_GFX_OVERLAY.
3. Build and install on an Ender 3 v2 or other printer with same screen (Icons updated on screen flash).
4. Enter menu "Configuration" -> "Probe Z Offset".

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Oct 16 2021 22:47:50)

### Printer model

Ender 2 v2

### Electronics

Stock electronics, rotated screen.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[image]

ellensp, 2021-10-20:
DWIN_MARLINUI_PORTRAIT mode isn't much better
[image]
Text to large, icons obscuring both ends of the number

aalku, 2021-10-20:
Looking at the photos we posted it seems like there is a limit based on portrait mode width.

ellensp, 2021-10-20:
If you cant read that it simply says 

edit Marlin/src/lcd/e3v2/common/dwin_api.cpp

above the line #include "dwin_api.h" incert the following
```CPP
#if ENABLED(DWIN_MARLINUI_LANDSCAPE)
  #define DWIN_WIDTH  480
  #define DWIN_HEIGHT 272
#endif

```

aalku, 2021-10-20:
I can read diff :) I don't think I can try it today. Thank you for your help!