MarlinFirmware/Marlin #22974: [FR] Adafruit featheresp32

mjmeans, 2021-10-17:
### Are you looking for hardware support?

'feather32esp" is a supported env in PlatformIO.

### Describe the feature you want

'feather32esp" is a supported env in PlatformIO. Is it possible to build Marlin for this board? What motherboard could work with it? I am only needed X/Y, but not Z since my project would be for like a pencil drawing platform and powered by an 18650 battery pack.

mjmeans, 2021-10-17:
I've tried adding this to esp.ini:

```
[env:featheresp32]
platform               = espressif32@2.1.0
extends                = env:esp32
board                  = featheresp32

```
And setting BOARD to `BOARD_ESPRESSIF_ESP32`. It compiles but will not upload to the HUZZAH32.

ellensp, 2026-10-30:
I've put together a branch with an Adafruit HUZZAH32 ESP32 Feather board and an X/Y plotter configuration: https://github.com/ellensp/Marlin/tree/adafruit-feather-esp32

You can inspect the changed files here: https://github.com/ellensp/Marlin/commit/e10fc3f4b8

Build it with PlatformIO using `env:adafruit_feather_esp32`. The configuration is already set up for two axes, with no Z, extruder or heaters, a servo for the pen, and SD card support.

Wiring (Feather header pins):

| Function | GPIO (label) |
|---|---|
| X step / dir | 26 (A0) / 25 (A1) |
| Y step / dir | 27 / 33 |
| X + Y enable (shared) | 32 |
| X / Y endstop | 34 (A2) / 39 (A3) |
| Pen servo | 14 |
| SD card SCK / MOSI / MISO / CS | 5 (SCK) / 18 (MO) / 19 (MI) / 21 |

- GPIO 34 and 39 have no internal pull-ups, so the endstop switches need external pull-up resistors (e.g. 10k to 3.3V).
- Pen up/down: `M280 P0 S<angle>`.
- Set `DEFAULT_AXIS_STEPS_PER_UNIT` and the bed size in `Configuration.h` for your mechanics.

This was tested on an ESP32 devkit with the same ESP32-WROOM-32 module, not on a Feather itself.

Since the Feather is a general-purpose development board rather than a printer controller, the pin layout above is just one of many possible choices, and it's tailored to this plotter. That makes it a bit too specific to add to Marlin itself, so it lives on this branch instead. I hope it gets your project moving!

closed 2026-10-30 by ellensp