MarlinFirmware/Marlin #22979: [FR] (PT100 with MAX31865 compatibility for 3 thermistors)

kurtis-potier-geofabrica, 2021-10-18:
### Is your feature request related to a problem? Please describe.

I am trying to integrate a Dyze Design Pulsar. It uses 3 PT100 thermistors for one extruder. I then will use the MAX31865 board to convert to SPI so I can read them (-5 selected as thermistor). The only is issue is that Marlin does not support 3 PT100. 

### Describe the feature you want

After some work, I have gone through conditionals_post.h, temperature.cpp, and other files to make an option for a third PT100 successfully build. I have not tested on hardware yet. It would be helpful if  a developer could properly add this so it is supported in future updates. Thank you!

kurtis-potier-geofabrica, 2021-11-17:
Update, I did get this to work correctly after some work. All thermal features appear to working correctly with this after using the machine for a week or so. I will take some time to formalize this and submit changes.

kurtis-potier-geofabrica, 2021-11-18:
It looks like conditionals_post.h has been cleaned up quite a bit since I cloned on 9/24. Will still try to revisit in the future, but timeline may be further out.

thisiskeithb, 2024-01-02:
You can now use MAX31865 for sensors 0-2 and bed.

https://github.com/MarlinFirmware/Marlin/blob/f02fa6339f003a8a5074131b50e0905f102ee8e4/Marlin/Configuration.h#L516

closed 2024-01-02 by thisiskeithb