MarlinFirmware/Marlin #22993: [BUG] "FileNotFoundError: [Errno 2] No such file or directory" when building LPC1768 in version 2.0.9.2

T-SquaredProductions, 2021-10-20:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I've recently gotten back into refurbishing my 3D Printer, but I've run into a slight problem. The system would not correctly home. The Y-Axis (table) would not activate, and it's been about year since I last touched my printer. Although, I was able to get the table to move using the "Move Axis" feature. I figured maybe the firmware on the board got corrupted, so I decided to reinstall it with the MacOS version of VSCode and PlatformIO.

However, when it came time to build the firmware, I got this error (I've replaced my user folder name with "ME" for privacy):

```
FileNotFoundError: [Errno 2] No such file or directory: '/Users/[ME]/Marlin':
  File "/Users/[ME]/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 168:
    env.LoadPioPlatform()
  File "/Users/[ME]/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "/Users/[ME]/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/pioplatform.py", line 102:
    board_config = env.BoardConfig()
  File "/Users/[ME]/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "/Users/[ME]/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/pioplatform.py", line 46:
    with fs.cd(env.subst("$PROJECT_DIR")):
  File "/Users/[ME]/.platformio/penv/lib/python3.9/site-packages/platformio/fs.py", line 37:
    os.chdir(self.new_path)
========================== [FAILED] Took 0.39 seconds ==========================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:00.390
==================== 1 failed, 0 succeeded in 00:00:00.390 ====================
The terminal process "platformio 'run'" terminated with exit code: 1.
```

I figured it was a bad folder name, so I took off the "2.0.x" off the end. That didn't solve the problem.

I'm not sure what is going on, but I never modified the platformio.ini file past inputting the environment variable, nor did I modify the config files past checking my old config files and copying its values for consistency.

### Expected behavior

I had expected to be able to build the binary firmware file without trouble. Nothing seems out of the ordinary.

### Actual behavior

The terminal threw up an error and stopped, even when I tried to change the folder name so there was no ambiguity. (i.e. a character in a folder name being changed for formatting purposes.)

### Steps to Reproduce

1. Configure Marlin using VSCode and PlatformIO.
2. Configure the platformio.ini file to use the LPC1768 environment.
3. Try to build the firmware.

### Version of Marlin Firmware

2.0.9.2

### Printer model

Creality C10 S5

### Electronics

BTT SKR 1.4

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Config & PlatformIO INI.zip](https://github.com/MarlinFirmware/Marlin/files/7377959/Config.PlatformIO.INI.zip)

ellensp, 2021-10-20:
But that is not this error. this error is something to do with platformio on your conputer
delete you .platformio in your home directory and try again

T-SquaredProductions, 2021-10-20:
> But that is not this error. this error is something to do with platformio on your conputer delete you .platformio in your home directory and try again

Do you mean the platformio.ini file for Marlin, or reinstall PlatformIO alltogether?