MarlinFirmware/Marlin #22996: [BUG] Printer doubling movement after print start

boxburn, 2021-10-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After homing all axis I can move the axis correctly in any direction both through Pronterface and the attached display. Once I try to print, every axis movement is doubled so a 20x20mm cube tries to print a 40x40mm cube. The print object located in the centre of the print bed will then try to print in the far corner due to the doubled movement. Pausing the print and manually moving the axis will then have doubled movement and it only goes back to moving correctly after pressing the printers reset button.

### Expected behavior

Move by Xmm at all times

### Actual behavior

Moves Xmm after homing and prior to print, moves Xmm x2 once print has started.

### Version of Marlin Firmware

2.0.x

### Printer model

Wanhao Duplicator i3

### Electronics

BigTreeTech SKR-2

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

dydyamotya, 2021-12-02:
Have you solved the issue? I have the similar config, as yours. SKR V2 with TMC2209 on all axes. But for me it is only a trouble on E axis. After nice two prints, suddenly extruder starts to eat double load of plastic completely destroying a model. I have REPRAP display and TMC2209 drivers set in config.

boxburn, 2021-12-02:
Yes, by disabling the display and setting the steppers to TMC2209_STANDALONE. Since then it has behaved on the double movements issue however, I cannot get it to print without very bad layer shifts. No matter what I do it will randomly skip on both the X and Y axis but that's not an issue to try and fix here.

dydyamotya, 2021-12-03:
> Yes, by disabling the display and setting the steppers to TMC2209_STANDALONE. Since then it has behaved on the double movements issue however, I cannot get it to print without very bad layer shifts. [...]

Okey, thank you. I'll try to figure out, what's the problem using debugging. Will write if I find out something.

closed 2022-04-01 by github-actions[bot]