MarlinFirmware/Marlin #23015: [BUG] Issues in "Level Corners" (config.h  vs. G35 Tramming Assist (config_adv.h)

sbaeder, 2021-10-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

It seems that in the current bugfix and 2.0.9.2 release, there is confusion in  the settings of these options - I didn't change anything in the configs in this area as I merged over changes from previous builds.  However, text in the menu changed - yes, see the pull request for that.  And in trying to dig further, the tramming code was in the config_adv.h for some time to turn on the G35 command.

But the tramming menu (now called that) in the LCD due to the Level Corners setting in the config.h went into an infinite loop, since the assisted stuff (G35 in config_adv.h) wasn't turned on.

Needs clean up so that duplicated information isn't mixed and conflated together.   I'm guessing a change top the configs in the configuration repo should also be done.

### Bug Timeline

2.0.9.2

### Expected behavior

just change text  based on the change notice

### Actual behavior

infinite loop around the periimiter

### Version of Marlin Firmware

2.0.9.2

### Printer model

home built

### Electronics

skr 1.3

Sonicboom247, 2022-04-15:
i wish i had the programming knowledge to rework this as it is an absolute nightmare to configure.  I / we should not have to spend hours trying to wrap our heads around the tramming points and how insanely complicated it is to figure out when you add the probe offsets, probing margins, min max bed size and the fact that that there are 4 tramming points labeled 1-4 yet the errors start at tramming point 0...
TRAMMING_POINT_XY point 0 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.
so is 0 actually 1?

with a 300x300 bed and a nozzle to probe offset of 30, -6, -2 i would imagine points
 { 0, 35 }, { 235, 35 }, { 235, 285 }, { 0, 285 } } are reachable but no matter what i get errors unless i disable tramming all together

thinkyhead, 2022-04-20:
Welcome to the topic @Sonicboom247. Please provide your configuration files for `bugfix-2.0.x` and we'll try to fix them or to fix the bug, as needed.

thinkyhead, 2022-04-21:
I'm happy to hear you got it sorted. The nozzle-to-probe offsets can definitely make it a little trickier.

closed 2022-04-21 by thinkyhead