MarlinFirmware/Marlin #23043: [BUG] LPC176X Neopixel timing bug

Lostish, 2021-10-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Neopixels efter 4'th pixel is random rainbow color.

After activated software SPI the neopixels acts in a random differend colors. Like a rainbow.
If i turn software SPI off it works as intended

Think is is the same as:  **[BUG] LPC1768: Incorrect Neopixel WS2812B timing in library #13997** 
But i dont know how to implement it or where to put the code in that BUG REPORT.

### Expected behavior

Same selected color all over the strip

### Actual behavior

Rainbow pattern in Neopixels after the 4'th pixel

### Steps to Reproduce

Turn on Neopixel
Acitvate software SPI

### Version of Marlin Firmware

Bugfix-2.0.x

### Printer model

Creality Ender 3

### Electronics

BTT SKR 1.4 Turbo

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7446522/Configuration.zip)

ellensp, 2021-10-30:
@p3p could you take a look...  I expect its cpu overload...

Lostish, 2021-10-30:
I have a second printer, CR10S.

Tested the same build and it works with Neopixel as is should.
The difference is the display attached.

CR10S has REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, works as intended.

Ender 3 has MKS_MINI_12864, and the leds are all rainbow.

Updated the issue with information.
And hope that will aid to the troubleshooting.

p3p, 2021-10-30:
@ellensp I would be surprised if its overloaded, it will be the interaction between multiple timing critical peripherals, Neopixels being the least important device in a printer but requiring the very high priority signal generation (disabled interrupts) to maintain the ns timing requirement.

thisiskeithb, 2021-10-31:
Adding a resistor & capacitor is recommended on BTT boards (probably others as well) when using NeoPixels.

I use [this little NeoPixel board](https://www.printedsolid.com/products/skr-neopixel-breakout-board) to clean things up.

descipher, 2021-11-02:
Have you followed this:
https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices