MarlinFirmware/Marlin #23044: [BUG] Fysetc S6 Y-Stepper Pins Incorrect

dappydarwin, 2021-10-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling either Bugfix-2.0.x or 2.0.9.2 the PINS file **pins_FYSETC_S6.h** section #if HAS_TMC_UART contains the wrong pin (PE13) for #define Y_SERIAL_RX_PIN when it should be PC4.  See S6 Schematic, page 7 Grid area C2 at https://github.com/FYSETC/FYSETC-S6/blob/main/hardware/V2.0/S6%20V2.0%20SCH%20.pdf .  

### Bug Timeline

at least a week ago

### Expected behavior

TMC communication should happen after startup for all drivers

### Actual behavior

TMC Communication Error for Y Stepper only.  M122 receives "Testing Y connection... Error: All LOW"

### Steps to Reproduce

Flash firmware with stock pins files
Reboot
Get TMC Connection Error displayed - M122 shows Testing Y connection... Error: All LOW

### Version of Marlin Firmware

v2.0.x-Bugfix distributed 2021-10-30 and v2.0.9.2

### Printer model

Monoprice MP10 Mini

### Electronics

Fysetc S6

### Additional information & file uploads

I moved the pin assignments from pins_FYSETC_S6_V2_0.h to the pins file called at the bottom (#include...), pins_FYSETC_S6.h.  This was just so I could keep the schematic comparison easier dealing with only one pins file.  This isn't necessary for a proper fix, just correcting the pin in pins_FYSETC_S6_V2.0.h will do.  After fixing, motion direction and Steps/mm are correct.
[Debug.zip](https://github.com/MarlinFirmware/Marlin/files/7447066/Debug.zip)

ellensp, 2021-11-01:
I cant fault your Y_SERIAL_RX_PIN PC4 from the diagram
But that creates a conflict with #define E1_SERIAL_TX_PIN PC4 in marlin
you also have  #define E1_SERIAL_TX_PIN                    PC5
but that is also in conflict with marlin #define E1_SERIAL_RX_PIN                PC5  which seem correct from the diagram.
 E1_SERIAL_RX_PIN                PC5  seem correct from the diagram.

So so me this means #define E1_SERIAL_TX_PIN  is not PC4 or  PC5

PC5 is USART-RX3 so we are looking for USART-TX3

Unfortunately there are 3  USART-TX3's listed
PB10 => SD_DETECT_PIN
PC10 => LCD_RESET_PIN
PD8 => Y_STEP_PIN

So i'm at an impasse

maralb1970, 2021-11-01:
I am running a Hiprecy Leo with a Fysetc S6 V2.0 controller. The Leo's are running a customized Marlin V2.0.5.1 Marlin Firmware. The Leo's came originally with a Fysetc F6 8 bit controller. The S6 is an upgrade for the Hiprecy Leo printers. Hiprecy/Fysetc provided a pins_Fysetc_S6_V2_0.h to run the S6 V2.0 with Marlin V2.0.5.1. My printer is running fine with it. Indeed, in this file Y_SERIAL_RX_PIN is assigned to PC4. I include the Hiprecy S6 V2 pins.h......maybe it helps
[pins_FYSETC_S6_V2_0.zip](https://github.com/MarlinFirmware/Marlin/files/7452821/pins_FYSETC_S6_V2_0.zip)
.

ellensp, 2021-11-01:
Tracing the gerber files E1_SERIAL_TX_PIN leads to PE5 (if I haven't lost the track)

ellensp, 2021-11-01:
I see what is going on here..
there is only one io line for tx/rx
Marlin/src/pins/stm32f4/pins_FYSETC_S6_V2_0.h defines
```
#if HAS_TMC_UART
  #define X_SERIAL_TX_PIN                   PE8
  #define Y_SERIAL_TX_PIN                   PC4
  #define Z_SERIAL_TX_PIN                   PD12
  #define E0_SERIAL_TX_PIN                  PA15
  #define E1_SERIAL_TX_PIN                  PC5
  #define E2_SERIAL_TX_PIN                  PE0
#endif

```
Then includes Marlin/src/pins/stm32f4/pins_FYSETC_S6.h
which sets the RX pins
```
    #define X_SERIAL_RX_PIN                 PE8
    #define Y_SERIAL_RX_PIN                 PE13
    #define Z_SERIAL_RX_PIN                 PD12
    #define E0_SERIAL_RX_PIN                PA15
    #define E1_SERIAL_RX_PIN                PC5
    #define E2_SERIAL_RX_PIN                PE0
  ```
The only other board that incudes pins_FYSETC_S6.h pins_FYSETC_SPIDER.h
And it sets all TX and RX pins so the ones in  pins_FYSETC_S6.h are ignored
So it is safe up update pins_FYSETC_S6.h 
From
```
  #ifndef Y_SERIAL_RX_PIN
    #define Y_SERIAL_RX_PIN                 PE13
  #endif

```
To
```
  #ifndef Y_SERIAL_RX_PIN
    #define Y_SERIAL_RX_PIN                  PC4
  #endif

```

dappydarwin, 2021-11-01:
Sorry, just back from work or I would have saved you a bit of hunting.
Yes, that's all spot on!

ellensp, 2021-11-01:
This has been corrected

closed 2021-11-01 by ellensp