MarlinFirmware/Marlin #23045: GT2560 rea A+ and Fysetc 12864 Mini 2.1 display

donytech, 2021-10-30:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

by enabling:
#define FYSETC_MINI_12864_2_1

the display doesn't work.
neither the neopixel led.

### Version of Marlin Firmware

2.0.9.2

### Printer model

Geeetech i3 Pro B

### Electronics

GT2560 rev A+

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Additional information & file uploads

The display works with this test code

[Fysetc_Mini_12864_Test.zip](https://github.com/MarlinFirmware/Marlin/files/7447418/Fysetc_Mini_12864_Test.zip)

ellensp, 2021-10-31:
I can not replicate this issue

I don't have a GT2560 rev A+, but its just a mega2560 really
I wired up the Fysetc 12864 Mini 2.1 display to the same pins as the GT2560 uses on my mega

```
> PIN:  16   Port: H1        LCD_RESET_PIN                          Output = 1 
> PIN:  17   Port: H0        DOGLCD_CS                              Output = 1  
> PIN:  18   Port: D3        BEEPER_PIN                             Output = 0  
> PIN:  19   Port: D2        BTN_ENC                                Input  = 1  
> PIN:  20   Port: D1        DOGLCD_A0                              Output = 1  
> PIN:  21   Port: D0        NEOPIXEL_PIN                           Output = 0  
> PIN:  38   Port: D7        SD_DETECT_PIN                          Input  = 1  
> PIN:  40   Port: G1        BTN_EN2                                Input  = 1  
> PIN:  42   Port: L7        BTN_EN1                                Input  = 1  
> PIN:  51   Port: B2        AVR_MOSI_PIN                           Output = 1  
> .                          DOGLCD_MOSI                            Output = 1  
> PIN:  52   Port: B1        AVR_SCK_PIN                            Output = 0  
> .                          DOGLCD_SCK                             Output = 0  
> PIN:  53   Port: B0        AVR_SS_PIN                             Output = 1  
> .                          SDSS                                   Output = 1  

```
[image]

Starting with stock Marlin config I only updated these settings
#define MOTHERBOARD BOARD_GT2560_REV_A_PLUS
#define SDSUPPORT
#define FYSETC_MINI_12864_2_1    // Type A/B. NeoPixel RGB Backlight
#define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
#define PINS_DEBUGGING

Which worked fine under Marlin

[image]

I then used your test program without making any hardware changes

It worked fine also.

[image]

ellensp, 2021-10-31:
And yes the Neopixeles works as expecting doing the NEOPIXEL_STARTUP_TEST in marlin on startup
The encoder and and even the sd card works as expected,

ellensp, 2021-10-31:
Oh and I wrote that Fysetc_Mini_12864_Test, a long time ago... https://github.com/ellensp/FYSETC-Mini-12864-tester

donytech, 2021-10-31:
I found the issue:
the default lcd contrast was 0 and not visible.
it was visible after command M250 c255 by pronterface.

Thank you friend!

closed 2021-10-31 by ellensp