MarlinFirmware/Marlin #23047: [BUG] Probe temperature compensation does not apply to Z Wizard, UBL, etc.

tombrazier, 2021-10-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Probe temperature compensation has effect in exactly one place: in G29 for `AUTO_BED_LEVELING_LINEAR` and `AUTO_BED_LEVELING_BILINEAR`.

I propose to submit a PR moving the compensation code into `probe.probe_at_point()` with an option disable PTC in a few specific instances.

The following would benefit from this:
- G29 for levelling algorithms `AUTO_BED_LEVELING_3POINT` and `AUTO_BED_LEVELING_UBL`. Also any future G29 algorithm anyone invents.
- The Z Probe Wizard: presently the Z Offset found by the wizard is incorrect unless the wizard is used at the temperatures defined by `PTC_PROBE_START`, `PTC_BED_START` and `PTC_HOTEND_START`.
- G33 - Delta Auto Calibration.
- The Tramming Wizard - if you take long enough using it for temperatures to change.
- DWIN_CREALITY_LCD_ENHANCED has a menu option that reports bed height.

In other cases it wouldn't make any difference one way or the other:
- G35 - Tramming Assistant.
- G34 - Z-Stepper automatic alignment.
- DWIN_CREALITY_LCD_JYERSUI has a menu option where Z heights are compared with each other.

In two cases, it makes sense for the user to specify whether to apply PTC:
- G30 - Single Z-Probe.
- M48 - Probe Accuracy Test.

Finally, G76 - PTC calibration needs to run without PTC being applied.

### Bug Timeline

It has always been there.

### Expected behavior

I expect UBL, for example, to apply PTC. And I expect the Z Probe Wizard to give me a correct value.

### Actual behavior

They don't.

### Steps to Reproduce

Enable `PROBE_OFFSET_WIZARD`.
Enable and calibrate PTC.
Use the Z Wizard at different temperatures: the result should be pretty much the same because of PTC but they aren't.

### Version of Marlin Firmware

Latest bugfix branch

### Bed Leveling

UBL Bilinear mesh

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7447652/configs.zip)

tombrazier, 2021-10-31:
So if G30 and M48 need an option to enable/disable PTC when probing, any suggestions for a parameter letter? P is already taken for M48. How about C which can take a 0 or 1 to disable or enable PTC respectively and defaults to 1?

tombrazier, 2022-02-14:
Just in case anyone wonders, I have not forgotten this. I am just getting inconsistent results in testing which I think is related to my machine's hardware. Working on it slowly.

thisiskeithb, 2022-02-22:
#23764 has been merged.

closed 2022-02-22 by thisiskeithb