MarlinFirmware/Marlin #23049: [BUG] Manualy moving Z through menu in 2:1 microstepping moves only every 0.2mm or 0.125mm

Battou2501, 2021-10-31:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I decided to text 2:1 microstepping on my CoreXY for Z axis for position accuracy so every position is full step or between steps. I use 8mm pitch lead screw so Im getting 50 steps per mm or 0.02mm per step (half full step). But when I flashed new firmware and tried to move Z through menu using 0.1mm steps or 0.025mm steps it was not moving through every half-step.

In 0.1mm steps it moves only every other 0.1mm. In 0.025mm steps it moves only every 5th 0.025mm step.

Im using SKR1.4 Turbo with TMC 2208 an all motors. Interpolation for Z was disabled, StealthChop enabled. 

Marlin version 2.0.9.2

### Bug Timeline

2.0.9.2

### Expected behavior

Z axis moves with 50 steps per mm, 0.02mm per step at 2:1 microstepping.

### Actual behavior

Only moves in 0.125mm or 0.2mm steps.

### Steps to Reproduce

In config set TMC2208 drivers, set Z axis microstepping to 2, disable its interpolation, try to move Z axis through menu.

### Version of Marlin Firmware

2.0.9.2

### Electronics

SKR 1.4 Turbo, TMC2208 on all steppers

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

ellensp, 2021-10-31:
// Moves (or segments) with fewer steps than this will be joined with the next move
#define MIN_STEPS_PER_SEGMENT 6

Battou2501, 2021-10-31:
> // Moves (or segments) with fewer steps than this will be joined with the next move #define MIN_STEPS_PER_SEGMENT 6

X_X Oh.... Thank you! Never new there was such a setting.... All these years using marling and never noticed it.(

closed 2021-10-31 by Battou2501