MarlinFirmware/Marlin #23052: [BUG] Y-axis grinds with TMC2209 sensorless homing

reidprichard, 2021-10-31:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I'm running an Ender 5 with the SKR v1.4 turbo board and TMC2209 drivers on X, Y, Z, and E0. I am using sensorless homing on X and Y. I was previously running Marlin 2.0.5.3 with everything working perfectly. However, after updating to 2.0.9.2, the Y axis will not home properly. With 99% of the things I have tried (including the exact settings I was previously using successfully), it grinds against the endstop until I power it down, it times out, or I give the carriage a tug. I have tried a plethora of tweaks to no avail: 
- Increasing homing current
- Decreasing homing current
- Increasing stall sensitivity to 255
- Decreasing stall sensitivity to 0
- Setting stall sensitivity to various intermediate values
- Increasing homing speed
- Enabling/disabling HOME_Y_BEFORE_X
- Enabling/disabling QUICK_HOME
- Enabling/disabling IMPROVE_HOMING_RELIABILITY

The only combination with any success was IMPROVE_HOMING_RELIABILITY with high homing current and high stall sensitivity, but it still would briefly grind when homing Y.

Here are my relevant settings that work with 2.0.5.3:
```
#if HAS_TRINAMIC_CONFIG

  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #if AXIS_IS_TMC(X)
    #define X_CURRENT       580        // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_CURRENT_HOME  200  // (mA) RMS current for sensorless homing
    #define X_MICROSTEPS     16    // 0..256
    #define X_RSENSE          0.11
    #define X_CHAIN_POS      -1    // <=0 : Not chained. 1 : MCU MOSI connected. 2 : Next in chain, ...
  #endif

  #if AXIS_IS_TMC(Y)
    #define Y_CURRENT       650
    #define Y_CURRENT_HOME  220
    #define Y_MICROSTEPS     16
    #define Y_RSENSE          0.11
    #define Y_CHAIN_POS      -1
  #endif

  #if AXIS_IS_TMC(Z)
    #define Z_CURRENT       580
    #define Z_CURRENT_HOME  Z_CURRENT
    #define Z_MICROSTEPS     16
    #define Z_RSENSE          0.11
    #define Z_CHAIN_POS      -1
  #endif

  #if AXIS_IS_TMC(E0)
    #define E0_CURRENT      650
    #define E0_MICROSTEPS    16
    #define E0_RSENSE         0.11
    #define E0_CHAIN_POS     -1
  #endif
#endif
```

```
#define SENSORLESS_HOMING // StallGuard capable drivers only

  #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY  106
    #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY  135
    //#define Z_STALL_SENSITIVITY  8
    //#define SPI_ENDSTOPS              // TMC2130 only
    //#define IMPROVE_HOMING_RELIABILITY
  #endif
```

### Bug Timeline

Observed soon after updating to Marlin 2.0.9.2 from 2.0.5.3

### Expected behavior

Sensorless homing should be smooth and quiet.

### Actual behavior

The Y axis grinds against the endstop.

### Steps to Reproduce

1. Install Marlin 2.0.9.2 with TMC2209 drivers.
2. Enable sensorless homing.
3. Compile and upload the firmware
4. Send G28.

### Version of Marlin Firmware

2.0.9.2

### Printer model

Creality Ender 5

### Electronics

SKR v1.4 Turbo, TMC2209 drivers

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[Marlin 2.0.9.2 config.zip](https://github.com/MarlinFirmware/Marlin/files/7449521/Marlin.2.0.9.2.config.zip)

Tannoo, 2021-11-02:
What does M914 report?

Tannoo, 2021-11-02:
I am also running an SKR v1.4 turbo board with TMC2209 drivers on X, Y, Z, and E0. I am also using sensorless homing on X and Y. I am currently running Marlin 2.0.9 (bugfix) with everything working perfectly.

As soon as I get my UBL matrix recorded on paper, I will be upgrading to 2.0.9.3-bugfix.

I don't have any issues with homing using these settings for my machine.

```c++
  #if AXIS_IS_TMC(X)
    #define X_CURRENT       600        // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_CURRENT_HOME  X_CURRENT  // (mA) RMS current for sensorless homing
    #define X_MICROSTEPS     16        // 0..256
    #define X_RSENSE          0.11
    #define X_CHAIN_POS      -1        // -1..0: Not chained. 1: MCU MOSI connected. 2: Next in chain, ...
    //#define X_INTERPOLATE  true      // Enable to override 'INTERPOLATE' for the X axis
  #endif
```
and

``` c++
  #if AXIS_IS_TMC(Y)
    #define Y_CURRENT       650
    #define Y_CURRENT_HOME  Y_CURRENT
    #define Y_MICROSTEPS     16
    #define Y_RSENSE          0.11
    #define Y_CHAIN_POS      -1
    //#define Y_INTERPOLATE  true
  #endif
```
and

```c++
    #define X_STALL_SENSITIVITY  106
    #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY  135
    #define Y2_STALL_SENSITIVITY Y_STALL_SENSITIVITY
```

closed 2021-11-03 by thisiskeithb