MarlinFirmware/Marlin #23053: [BUG] (E0 on BTT Octopus won't extrude)

kj4sst, 2021-10-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Attempting to get a switching nozzle dual extruder working. The machine is core-xy, 2 extruders, 4 Z steppers. Marlin compiles E0 does not work. X/Y = motors 0/1, Z = motor 2, Z2 = motor 3, E0 = motor 4 (this is the first automatic assignment), E1 = motor 5, Z3 = motor 6, Z4 = motor 7. When M17 is sent all of the steppers enable except for E0. It appears to be a problem with how the steppers are being automatically assigned somewhere in the compile. I have completed these test with the latest release and bugfix.

Note: I also compiled the same configuration with NO EXTRUDERS and Z3 (the first automatically assigned stepper) fails to enable with M17 just like E0 did in the above configuration.

### Expected behavior

I expected both stepper E0 and E1 to function.

### Actual behavior

Only stepper E1 functions.

### Steps to Reproduce

1. Use the included config files.
2. Compile Marlin.
3. Upload to an Octopus board with 8 steppers.
4. Send command M17 to test steppers.

### Version of Marlin Firmware

2.0.x-bugfix(10/31/21)

### Printer model

Custom

### Electronics

BBT Octopus, TMC2209 drivers

### Bed Leveling

UBL Bilinear mesh

### Additional information & file uploads

The pin file for the octopus board is included as for the switching nozzle head being used to work 2 servos are needed to be defined. 
The only change in the pins file is adding a copy of the servo define making it as follows:
'''
#define SERVO0_PIN                        PB6
#define SERVO1_PIN                        PB6
'''
[MarlinConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/7449724/MarlinConfigFiles.zip)

thisiskeithb, 2021-10-31:
@thinkyhead Should we remove the defined defined Z2 drivers in the Spider & Octopus boards and define them as extruder drivers (without a hotend pin paired with them)?

They seem to mess with the auto-assignments & these types of reports keep popping up.

thinkyhead, 2021-10-31:
Yes, unless the board comes with "Z2" specifically, with no heater/thermistor pins that pertain to the extra motor so there's no possibility to use it as an E axis. We should avoid defining X2, Y2, Z2, etc., because it reduces the versatility of the board. Then, the configurations that need a Z2 should be set up so that a Z2 will get auto-assigned, if possible.

thisiskeithb, 2021-10-31:
I'll put in a PR after some more config building/cleanup I need to wrap up.

ellensp, 2021-10-31:
If there is an issue here we need to find it, as FYSETC Spider does have pins labeled Z2 on the controller so renaming them is not an option

thisiskeithb, 2021-10-31:
> If there is an issue here we need to find it, as FYSETC Spider does have pins labeled Z2 on the controller so renaming them is not an option

Oops. I thought they were labeled MOT0-7 like the Octopus. Looks like more work needs to be done around auto-assignment of pins then.

exxosuk, 2022-02-03:
Is there a fix for this ?  I cannot get E0 working, but XYZ work fine (extruder motor and cables work in the X-driver connector).

SpectrumTech, 2022-02-16:
What kind of power supply do you have?

I have an issue with a BTT SKR E3 Turbo board (same stepper drivers) that has never been resolved that is similar in nature. When printing, E0 will not extrude. I have 2 Z steppers to use auto Z align, which is really kind of needed on my CR10S5. I have the same board and an identical setup in my ender 3 and it works fine, auto z align works, it extrudes, prints great. The glaring difference is that the ender 3 is 24V and the CR10S5 is 12V. I don't think the TMC2209, marlin and a 12V PS work together. It's the only thing I can figure. Is there voltage setting somewhere that I am missing? I have CHOPPER_TIMING set to CHOPPER_DEFAULT_12V. I have tried turning Steath Chop on and off, I have tried turning square wave on, I have messed with motor currents and any and all combinations make the printer sound like it's going to come apart and leaving everything above enabled is when most everything works, except it will not extrude...

kj4sst, 2022-02-16:
I am using a 24V PSU to power the board and steppers.  As far as I can tell, the issue in my case is caused by the steppers being assigned incorrectly, skipping one of the positions, when compiling marlin.

kj4sst, 2022-02-25:
After more debugging, I discovered that the enable pin on MOTOR4 was not being pulled low on the board, preventing the stepper from working. Getting a new board confirmed this as now all of the steppers work correctly.

closed 2022-02-25 by kj4sst

jmger, 2022-04-23:
Hi I am having the same issue. Everything is working on my Octopus pro 429 but the E0 Extruder does not work. I am not sure I know how to perform the steps you guys are taking about Any help will be appreciated. I am using Marlin 2.0.9.3

thisiskeithb, 2022-04-23:
@jmger: This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

jmger, 2022-04-29:
> Yes, unless the board comes with "Z2" specifically, with no heater/thermistor pins that pertain to the extra motor so there's no possibility to use it as an E axis. We should avoid defining X2, Y2, [...]

Hi, I have the Octopus Pro 429 board. I wanted to try your solution. Can you explain how to do this a little more detail and or provide code section. I have just disabled all ics/ extruders.etc that I am not using in both the Configuration.h file and the Octopus_pin....h file. so far. I have not defined an actual Z2 extruder section. that's the Code I am looking for. I tried other things like switching E0 and E1 in pin.h files, etc. with no change. Thank You