MarlinFirmware/Marlin #23063: [FR] M150 behaviour with two neopixel strips and NEOPIXEL2_SEPARATE

sblaisot, 2021-11-02:
### Is your feature request related to a problem? Please describe.

When using two neopixel stripes with `NEOPIXEL2_SEPARATE` (independant led stripes), `M150 S0...` let you command first led stripe and `M150 S1 ...` allow you to command second led stripe.

Before #22174 `M150` without `S` parameter was used to command only first stripe. After that Pull Request, `M150` without an S parameter command both stripes.

### Describe the feature you want

As far I understand, It could be nice to command both stripes with a single command (for example if you use two stripes, one for each side of the frame/chassis). Pinging @GeoSaffer here as author of #22174, maybe he can share his use case here for having `M150` command both stripes.

However, in some other cases, like in my setup, you can use the two stipes in totally unrelated places, like the first one on the printer frame and the second one on the head, so switching them both doesn't make much sense. in addition, somme  "clients" do not handle two stripes and use hardcoded `M150` command without `S` parameter to manage leds like BTT touchscreens. In such client, you can't use your two stripes separately if `M150` command the two stripes.

So it seems desirable to make this behaviour configurable, adding something like a `NEOPIXEL2_M150_COMMAND_BOTH` config parameter to configure if `M150` command without `S` parameter should command both stripes or only the first one.

I'm opening this Feature Request for discussion before submitting a PR.

ellensp, 2021-11-02:
see     //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel

sblaisot, 2021-11-02:
@ellensp you didn't get it. I'm aware of `NEOPIXEL2_INSERIES`. I'm talking about `NEOPIXEL2_SEPARATE` which is the opposite.

and about the behaviour of `M150` without `S` parameter in separate mode.

sblaisot, 2021-11-02:
The question can be shortened as "should M150 without S parameter be treated like S0 or like S0 + S1" ?

I'm saying that there is no universal answer and this should be configurable, and asking for comment before submiting a PR.

ellensp, 2021-11-02:
Without S it should control both
With S0, string 1
with S1 string 2
I really don't see the point of this.. but reopening against my better judgment.
Whats one more FR that no one else will ever want in the list of 400 other FR's

sblaisot, 2021-11-02:
the point is that some "clients" like BTT TFT screens have a led control menu using M150 without S parameter to be compatible with a maximum of configuration (from a single dump led stripe to multiple neopixel). It has been built with a single stripe setup in mind and when havong two neopixel stripes it command both, which is anoying.

and that's not a feature request nobody would want, because I need it and propose to turn it into a PR if we come to an agreement on that. Otherwise, I will need to maintain it as a patch outside of marlin.

I really don't get the point of changing the original behaviour (control first stripe only) to controlling both stripes without notice (no changelog entry specifically), even if this parameter is undocumented (no `S` parameter in `M150` documentation)

closed 2022-12-07 by thinkyhead