MarlinFirmware/Marlin #23065: Defining and Using Variables in GCode

jweiss2, 2021-11-02:
### Is your feature request related to a problem? Please describe.

I am relatively new with Marlin and GCode, but I tried incorporating variables in my GCode (as described here: https://gcodetutor.com/cnc-macro-programming/cnc-variables.html) without success. I searched in the forums and found a post asking for this feature about 10 years ago with no update. I am just trying to optimize a value for a very repetitive movement, and being able to define a variable would save a lot of time when troubleshooting.

Thanks!

### Describe the feature you want

Support for defining and using variables in GCode as described here: https://gcodetutor.com/cnc-macro-programming/cnc-variables.html

DerAndere1, 2021-11-03:
The following is not a solution for existing gcode that includes variables, but it can serve as a workaround (or even a more powerful alternative) for those who are willing to learn Python basics: Use printcore (comes with printrun) as described at https://github.com/kliment/Printrun/blob/master/README.md#using-macros-and-custom-buttons or the python package [mecode](https://github.com/jminardi/mecode) to generate/send gcode

stevekenney318, 2022-10-19:
I would also love to see this feature added for use in Marlin gcode... I'm really quite surprised it's not there. 

Thanks, Steve

DerAndere1, 2024-07-27:
Also interesting: a library that can be used by g-code senders to post-processing LinuxCNC G-code to expand variables and o-codes. And there is an old pull request which is work in progress: https://github.com/MarlinFirmware/Marlin/pull/24654