MarlinFirmware/Marlin #23067: [BUG] Ender 3 with FYSECT Cheetah board, screen blank

J0hnMatrix, 2021-11-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After configuring the 2.0.9.2 version of Marlin, screen is blank and no response from host or anything.
Attached, the 2.0.9 working build and 2.0.9.2 not working one.
Bug is still present on 'bugfix-2.0.x'

### Expected behavior

Marlin boot up and works normally

### Actual behavior

Screen is blank

### Steps to Reproduce

Power on the printer

### Version of Marlin Firmware

2.0.9.2

### Printer model

Creality Ender 3

### Electronics

FYSETC Cheetah board

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin-2.0.9_en_OK.zip](https://github.com/MarlinFirmware/Marlin/files/7466506/Marlin-2.0.9_en_OK.zip)
[Marlin-2.0.9.2_en_NOK.zip](https://github.com/MarlinFirmware/Marlin/files/7466508/Marlin-2.0.9.2_en_NOK.zip)

arifr007, 2021-11-04:
have you try reverse the pin?

J0hnMatrix, 2021-11-04:
Which one? and why have it been modified?

closed 2022-01-18 by github-actions[bot]

thinkyhead, 2022-04-21:
@J0hnMatrix — Here are your configurations updated for the latest `bugfix-2.0.x`. Hopefully the issue has been solved, but if not then we can investigate further. Whenever you have anything to report, please add your comments to #23641.

[marlin-cheetah.tar.gz](https://github.com/MarlinFirmware/Marlin/files/8529070/marlin-cheetah.tar.gz)

thinkyhead, 2022-04-21:
Duplicate of #23641