MarlinFirmware/Marlin #23068: [BUG] M600 - too many arguments for format [-Wformat-extra-args]

rubienr, 2021-11-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When building Marlin firmware i see a compiler warning `too many arguments for format [-Wformat-extra-args]` caused in `Configuration.h` by `#define FILAMENT_RUNOUT_SCRIPT "M600"`. Modifying the code to `#define FILAMENT_RUNOUT_SCRIPT "M600 T%c"` seems to satisfy the compilation.

### Bug Timeline

seen first time about 2021.03.01

### Expected behavior

No compiler warning.

### Actual behavior

Compiler warning:

````
pio run --environment BIGTREE_SKR_PRO
...
    Marlin/src/feature/runout.cpp: In function 'void event_filament_runout(uint8_t)':
    Marlin/src/feature/../inc/../../Configuration.h:1540:34: warning: too many arguments for format [-Wformat-extra-args]
 1540 |   #define FILAMENT_RUNOUT_SCRIPT "M600"
/home/rubienr/.platformio/packages/framework-arduinoststm32/cores/arduino/avr/pgmspace.h:71:40: note: in definition of macro 'sprintf_P'
   71 | #define sprintf_P(s, ...) sprintf((s), __VA_ARGS__)
      |                                        ^~~~~~~~~~~
Marlin/src/feature/runout.cpp:132:25: note: in expansion of macro 'PSTR'
  132 |       sprintf_P(script, PSTR(FILAMENT_RUNOUT_SCRIPT), tool);
      |                         ^~~~
Marlin/src/feature/runout.cpp:132:30: note: in expansion of macro 'FILAMENT_RUNOUT_SCRIPT'
  132 |       sprintf_P(script, PSTR(FILAMENT_RUNOUT_SCRIPT), tool);
      |                              ^~~~~~~~~~~~~~~~~~~~~~
...
````

### Steps to Reproduce

1. clone the upstream Git repository and switch to `bugfix-2.0.x` branch
2. replace the configurations with the attached one
3. `pio run --environment BIGTREE_SKR_PRO`

### Version of Marlin Firmware

Marlin bugfix-2.0.x Nov 3 2021

### Printer model

Creality Ender 5 Plus (heavily modified)

### Electronics

Upgrade Board: BTT SKR PRO

### Add-ons

-

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7468419/configuration.zip)

sjasonsmith, 2021-12-31:
I have also encountered this. We can't just change that default, because the format argument is only supplied if `MULTI_FILAMENT_SENSOR` is defined.

@thisiskeithb & @ellensp, I've prototyped a `static_assert` which will check for the inclusion (or absence) of `%c` in `FILAMENT_RUNOUT_SCRIPT`. Where do you think this should go?

There is some advantage in keeping constexpr functions and their static_asserts in CPP files so they only have to be executed once during compilation. I could put it in `Warnings.cpp` (even though it produces errors), or I could put it in `runout.cpp` where the runout script is actually consumed.

sjasonsmith, 2021-12-31:
Another option might be to disable that warning for that line of code...
I guess it depends on whether there are valid workflows where the runout script would not use the tool parameter.

This is ultimately more fragile the closer you look at it. For example, you can only use `%c` once in the runout script. If you tried to use it more than once you'd have trouble, although gcc might block compilation.

sjasonsmith, 2021-12-31:
> although gcc might block compilation.

Nope, the way the string is assembled doesn't allow GCC to check for this at compile time. Multiple `%c` formatters in the runout script is going to cause undefined behavior, although most likely just a garbage character in the command.

virus-sss, 2022-05-28:
Тоже столкнулся с этой проблемой при смене монитора на MKS tft 35

thisiskeithb, 2023-01-14:
Addressed in https://github.com/MarlinFirmware/Marlin/pull/25150

closed 2023-01-14 by thisiskeithb