MarlinFirmware/Marlin #23070: [BUG] Update to Marlin 2.0.9.2 fw always fails on Biqu SE-BX

digant73, 2021-11-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I know the support to extui is no more provided since the first (and only one working) 2.0.6.1 version provided here https://github.com/bigtreetech/BIQU-BX (last update by thisiskeithb) but after that 2.0.6.1 version I was never able to install any other Marlin fw version on Biqu SE-BX.
Basically after 100% of fw is loaded, the printer resets. After rebooting, it is displayed 100% is updated and after few seconds the printer resets again and so on indefinitely.
Attached my last fw (based on official 2.0.9.2) configuration files (all the changes I made are marked with comment "DIGA-TECH") and the changes are basically the same applied on 2.0.6.1 that was working.
Attached also a video of the problem.

### Expected behavior

fw update properly installed and printer usable

### Actual behavior

the printer continuosly reboots trying to install the fw update

### Version of Marlin Firmware

official 2.0.9.2

### Printer model

Biqu SE-BX

### Additional information & file uploads

Link to Marlin 2.0.9.2 configuration:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7469688/Marlin.zip)

Link to video:
https://www.youtube.com/watch?v=ng5jp30W-0s

thisiskeithb, 2021-11-03:
> last update by thisiskeithb

To clarify, my PRs were to add STLs of the 4040 drawer & parts fan duct, not the firmware.

I'll get my BX updated to the latest bugfix and see if I can reproduce the issue. In the meantime, have you tried the configs from https://github.com/MarlinFirmware/Configurations/pull/604?

bigtreetech, 2021-11-04:
https://github.com/stm32duino/Arduino_Core_STM32/blob/main/libraries/SoftwareSerial/src/SoftwareSerial.cpp#L186
Hello, Please delete or comment out here.
[image]

There is no time to dig deep into the cause now. You can fix it temporarily like this before finding the root cause

digant73, 2021-11-04:
@thisiskeithb yes, I also used the looxonline's last branch with the same result of my configuration.
@bigtreetech hmm, using vscode I see that this peace of code is already commented out in Marlin code (Marlin/src/HAL/STM32/SoftwareSerial.cpp)

digant73, 2021-11-05:
@bigtreetech with your suggestion it worked. I also verified that setting SERIAL_PORT different than 1 or 4 the fw doesn't boot (the screen remains black but there is no permanent reboot as before). SERIAL_PORT 1 is the one (as it was in 2.0.6.1) I used to connect the TFT35 V3 at 250000).
As far as I see the piece of code responsible for the problem is on the Arduino framework since some time now so I was expecting the same behavior on other printers

digant73, 2021-11-05:
Bug temporary fixed!

thisiskeithb, 2021-11-05:
> I used to connect the TFT35 V3 at 250000.

Which printer do you have? The BX comes with a TFT70 directly connected the board/MCU with a 40 pin cable and it doesn't run it's own firmware.

digant73, 2021-11-06:
yes, sure it has the TFT70 driven by mainboard (now with only Marlin mode GUI due no extui is currently available since 2.0.6.1). But I also connected on the mainboard's TFT connector a second TFT (BTT TFT35 V3) for testing purposes.
For this extra TFT, serial port 1 is used (as it was set by default on 2.0.6.1) at 250000 (I never had any issue in the past with that rate

[image]
).

thisiskeithb, 2021-11-15:
You can modify `stm32h7.ini` and swap out: https://github.com/MarlinFirmware/Marlin/blob/36b2650f65e036ce1b02da568364230201a22b80/ini/stm32h7.ini#L28

with: 
```ini
platform_packages  = framework-arduinoststm32@https://github.com/thisiskeithb/Arduino_Core_STM32/archive/refs/heads/biqu-bx-workaround.zip
```

This is based on rhapsodyv's STM32/USB fixes + the workaround mentioned by BigTreeTech above.

thisiskeithb, 2021-11-15:
Reopening so this can be tracked. I'll likely put in a PR to change the platform packages URL for the BX until BigTreeTech can look at it further.

thisiskeithb, 2021-11-23:
#23131 was merged.

rhapsodyv, 2022-07-03:
This is the right fix: https://github.com/stm32duino/Arduino_Core_STM32/commit/f4c200f9d7d1258a15e1a2c56dbdd9d370e1b5b9

`active_listener` sometime is null and it's missing a null check, fixed in that commit.

thisiskeithb, 2022-07-03:
> This is the right fix: https://github.com/stm32duino/Arduino_Core_STM32/commit/f4c200f9d7d1258a15e1a2c56dbdd9d370e1b5b9

Thanks for finding the proper fix!

I've updated my Arduino Core STM32 fork/BX workaround branch to backport this change/properly fix this bug: https://github.com/thisiskeithb/Arduino_Core_STM32/commit/8b3522051ac21809c4e76146bbbc546970271039

closed 2022-07-03 by rhapsodyv