MarlinFirmware/Marlin #23071: [BUG] BTT SKR V1.4 Turbo, BLTouch, TMC2209 Failing Firmware Compile "LPC1769 environment problem."

nathandtrone, 2021-11-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to compile bugfix-2.0(and tried the release version), for a BTT SKR V1.4 Turbo, with TMC2209v1.2's and BL Touch Sensor... I ported a copy of the example configuration(`Creality/Ender-5/BigTreeTech SKR E3 Turbo with BLTouch`) and updated it with the values I could manage(I'm replacing a dying MB for a TEVO Black Widow, which has hardly any support, so I decided to move on).

During my compile on both BugFix2.0 and the Most recent Release, I am getting this error. I saw an old bug thread but I didn't understand what they were saying, and doesn't look like there was a solution, or caused problems for them.

### Bug Timeline

new bug. I haven't been able to get a good build going.

### Expected behavior

I am trying to use Arduino(1.8.13) to compile a `firmware.bin` file to image my board. It should verify and complete, and then I would just select "Sketch > Export compiled Binary" and be done.

### Actual behavior

I've set the Board for "Arduino Mega or Mega 2560", Processor to "ATmega2560 (Mega 2560)" and made sure I had the libraries required for compiling. Hit verify/compile and receive the following errors:
```
In file included from c:\users\user\appdata\local\temp\arduino_build_150164\sketch\src\pins\lpc1768\pins_btt_skr_v1_4.h:28:0,
                 from c:\users\user\appdata\local\temp\arduino_build_150164\sketch\src\pins\lpc1769/pins_BTT_SKR_V1_4_TURBO.h:34,
                 from c:\users\user\appdata\local\temp\arduino_build_150164\sketch\src\pins\pins.h:417,
                 from sketch\src\inc/MarlinConfig.h:34,
                 from sketch\src\MarlinCore.h:24,
                 from sketch\src\MarlinCore.cpp:31:
c:\users\user\appdata\local\temp\arduino_build_150164\sketch\src\pins\lpc1768\env_validate.h:25:4: error: #error "Oops! Make sure you have the LPC1769 environment selected in your IDE."
   #error "Oops! Make sure you have the LPC1769 environment selected in your IDE."
    ^~~~~
In file included from sketch\src\inc/MarlinConfig.h:49:0,
                 from sketch\src\MarlinCore.h:24,
                 from sketch\src\MarlinCore.cpp:31:
sketch\src\inc/SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
    ^~~~~
In file included from sketch\src\inc/MarlinConfig.h:49:0,
                 from sketch\src\MarlinCore.h:24,
                 from sketch\src\MarlinCore.cpp:31:
sketch\src\inc/SanityCheck.h:81:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
    ^~~~~
In file included from sketch\src\inc/MarlinConfig.h:49:0,
                 from sketch\src\MarlinCore.h:24,
                 from sketch\src\MarlinCore.cpp:31:
sketch\src\inc/SanityCheck.h:2406:8: error: #error "Enable USE_XMIN_PLUG when homing X to MIN."
       #error "Enable USE_XMIN_PLUG when homing X to MIN."
        ^~~~~
In file included from sketch\src\inc/MarlinConfig.h:37:0,
                 from sketch\src\MarlinCore.h:24,
                 from sketch\src\MarlinCore.cpp:31:
c:\users\user\appdata\local\temp\arduino_build_150164\sketch\src\core\macros.h:218:48: error: pasting "ENA_" and "/* Give feedback on EEPROM commands. Disable to save PROGMEM.*/" does not give a valid preprocessing token
 #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
                                                ^
c:\users\user\appdata\local\temp\arduino_build_150164\sketch\src\hal\avr\timers.h:101:22: note: in definition of macro '_CAT'
 #define _CAT(a,V...) a##V
                      ^
c:\users\user\appdata\local\temp\arduino_build_150164\sketch\src\core\macros.h:218:44: note: in expansion of macro 'CAT'
 #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
                                            ^~~
c:\users\user\appdata\local\temp\arduino_build_150164\sketch\src\core\macros.h:225:35: note: in expansion of macro '_ENA_1'
 #define TERN0(O,A)          _TERN(_ENA_1(O),0,A)    // OPTION ? 'A' : '0'
                                   ^~~~~~
sketch\src\module/settings.h:45:13: note: in expansion of macro 'TERN0'
         if (TERN0(EEPROM_CHITCHAT, success)) report();
             ^~~~~
In file included from sketch\src\module/stepper/indirection.h:44:0,
                 from sketch\src\module/stepper.h:47,
                 from sketch\src\MarlinCore.cpp:51:
sketch\src\module/stepper/trinamic.h:29:10: fatal error: TMCStepper.h: No such file or directory
 #include <TMCStepper.h>
          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

```

### Steps to Reproduce

1. Download Bugfix2.0
2. Download Example Configurations
3. Use "Creality/Ender-5/BigTreeTech SKR E3 Turbo with BLTouch"
4. Install prereq libs
5. Compile

### Version of Marlin Firmware

bugfix-2.0.x 2021-11-03

### Printer model

TEVO Black Widow (Soon to be BTT SKR Black Widow)

### Electronics

BTT SKR V1.4 Turbo

### Add-ons

BL-Touch; TMC2209-V1.2; Heated Bed; 1 Extruder; 3 Axis Motors.

### Bed Leveling

UBL Bilinear mesh

### Host Software

SD Card (headless)

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7471700/Configuration.zip)

thisiskeithb, 2021-11-04:
[Use VScode + PlatformIO](https://marlinfw.org/docs/basics/install_platformio_vscode.html).

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-11-04 by thisiskeithb

ellensp, 2021-11-04:
The LPC1768/9 chipset is only compatible with platformio.  There is no way to compile this under the Arduino IDE