MarlinFirmware/Marlin #23074: [BUG] STM32F103RET6_creality 4.2.7 board ender 3 pro

ruck312, 2021-11-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Keep getting an error

### Expected behavior

Compile

### Actual behavior

Fails

### Version of Marlin Firmware

2.0.x

### Printer model

Ender 3 PRo

### Electronics

4.2.7 Board

### Additional information & file uploads

Building in release mode
Compiling .pio\build\STM32F103RET6_creality\src\src\inc\Warnings.cpp.o
Linking .pio\build\STM32F103RET6_creality\firmware-20211103-234110.elf
c:/users/johnny's/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file C:/Users/Johnnys/.platformio/packages/framework-arduinoststm32/system/ldscript.ld -Os -mthumb -mcpu=cortex-m3 --specs=nano.specs -Wl,--gc-sections,--relax -Wl,--check-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--defsym=LD_MAX_SIZE=524288 -Wl,--defsym=LD_MAX_DATA_SIZE=65496 -Wl,--defsym=LD_FLASH_OFFSET=0x7000 -Wl,--default-script C:/Users/Johnnys/.platformio/packages/framework-arduinoststm32/variants/MARLIN_F103Rx/ldscript.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RET6_creality\firmware-20211103-234110.elf] Error 1

ellensp, 2021-11-04:
Cannot reproduce this issue.
The script  pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py correctly copies buildroot/share/PlatformIO/variants/MARLIN_F103Rx into  .platformio/packages/framework-arduinoststm32/variants

ellensp, 2021-11-04:
issue may be your user name
in one part it says  c:/users/johnny's/ but when linking its C:/Users/Johnnys

closed 2022-01-14 by github-actions[bot]