MarlinFirmware/Marlin #23078: [BUG] Unable to compile for LPC1769 SKR V1.4 Turbo

Nesquirt, 2021-11-05:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I'm trying to build marlin fw for LPC1769, i've uncommented some row in according with the BTT tft35 e3 v3 github guide
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Before changing that row marlin was build succesfull, now i've some error 

**My configuration:**
[Marlin-2.0.x+skr1,4+lcd12864.zip](https://github.com/MarlinFirmware/Marlin/files/7479913/Marlin-2.0.x%2Bskr1.4%2Blcd12864.zip)

### Bug Timeline

Before adding tft35 e3 v3 lines, building is okay

### Expected behavior

Compile successfully

### Actual behavior

Compile unsuccessfully

### Steps to Reproduce

- Set default environment as LPC1769

- Select board BTT_SKR_V1_4_TURBO

- Select Marlin Auto Build

### Version of Marlin Firmware

2.0.91

### Electronics

skr 1.4 turbo and tft35 e3 v3 

### Additional information & file uploads

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.9
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.5.0
Building in release mode
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_MinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_pin_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\usb_serial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\watchdog.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\HAL_MinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm_arm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm_thumb.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarmbytab.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarmmem.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwinder.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\pause.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\tmc_util.cpp.o
Compiling .pio\build\LPC1769\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\LPC1769\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\LPC1769\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\LPC1769\src\src\gcode\config\M221.cpp.o
In file included from Marlin\src\gcode\calibrate\../../module/stepper/../../feature/../lcd/../feature/pause.h:37,
                 from Marlin\src\gcode\calibrate\../../module/stepper/../../feature/../lcd/marlinui.h:59, 
                 from Marlin\src\gcode\calibrate\../../module/stepper/../../feature/tmc_util.h:25,        
                 from Marlin\src\gcode\calibrate\../../module/stepper/trinamic.h:35,
                 from Marlin\src\gcode\calibrate\../../module/stepper/indirection.h:44,
                 from Marlin\src\gcode\calibrate\../../module/stepper.h:47,                               e.h:37,
                 from Marlin\src\gcode\calibrate\G28.cpp:27:
Marlin\src\gcode\calibrate\../../module/stepper/../../feature/../lcd/../feature/../libs/nozzle.h:24:10: fatal error: ../inc/MarlinConfig.h: No such file or directory
   24 | #include "../inc/MarlinConfig.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\gcode\calibrate\G28.cpp.o] Error 1                                        tal error: ../inc/MarlinConfig.h: No such file or directory
================================================================================================================= [FAILED] Took 9.45 seconds =================================================================================================================

Environment    Status    Duration                                                                         ======= [FAILED] Took 9.45 seconds ================================================================================================================= 
-------------  --------  ------------
LPC1769        FAILED    00:00:09.448
============================================================================================================ 1 failed, 0 succeeded in 00:00:09.448 ============================================================================================================PS E:\Simone\Documents\STL e SOLID\Tool stampante 3d\Marlin-2.0== 1 failed, 0 succeeded in 00:00:09.448 ============================================================================================================.x+skr1,4+lcd12864> echo "done" >"C:\Users\wilru\AppData\Local\Temp\ipc"                                  wilru\AppData\Local\Temp\ipc"
PS E:\Simone\Documents\STL e SOLID\Tool stampante 3d\Marlin-2.0.x+skr1,4+lcd12864>

ellensp, 2021-11-05:
Provided configs (extracted from the archive) compiles without issues on release 2.0.9.1

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-11-05 by ellensp

Nesquirt, 2021-11-05:
I've compiled before without issue other version,it can be a vs-code error?