MarlinFirmware/Marlin #23085: [BUG] CORE XY with SKR2 B and TMC2130 wont home in X or Y

Slipery, 2021-11-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

X and Y axis move as required in both X and Y direction from the front panel or from Pronterface but when Homing X or Y the head moves diagonally,  i.e only one motor turns, always the same motor, but in opposite direction for X and Y.

### Bug Timeline

This is on a SKGO where I have3 had to replace the SKR1.3 board with SKR 2b board. I have tried 2.0.9.1 2.0.9.2 and the latest bugfix.

### Expected behavior

When homing X I expect it to move to the left.
When Homing Y I expect it to move to the front.

### Actual behavior

Homing X it moves both to the left and to the rear.
Homing Y it moves both to the right and to the front.
However when moving either X or Y axis the movement is as expected.

### Steps to Reproduce

Power on printer
From the control Panel slect HOME X axis or HOME Y axis

### Version of Marlin Firmware

2.0.9.2 and latest bugfix

### Printer model

SKGO Core XY with SKR2 and TMC2130 drivers along with BLTouch

### Electronics

Control Board upgraded to SKR2B

### Add-ons

BLTouch, and TFT35v3

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/7489757/Configuration.files.zip)

Slipery, 2021-11-25:
Definitely a bug in either the TMC2130-SKR 2 -core XY set up.  Have upgraded the drivers to 2209 and updated marlin to reflect this and no longer a problem.  2130's however are still a problem.

gonzaloalbito, 2021-12-11:
I'mexperiencing similar issue with my new SKR mini E3 V3.0 on my ender-6.
Steps per mm are properly set, and kinematic is set to CoreXY but only X stepper is moving at proper speed. Y motor barely moves causing diagonal movent. It happens when homing and when explicitly moving axis in octoprint. I've also tested to flash it with cartesian kinematics with "same" results, having the X stepper moving at full speed when moving the X axis but the Y stepper moving too slow when moving the Y axis... I've also tested the pre build ender-5 firmware from BTT SKR mini v2 and it seems both axis are moving fine but i cannot use that because it's a cartesian printer.

larquetoux, 2021-12-13:
Hello, I have a SKR 2-B board with TMC 5160 on homemade printer Core XY.

I had the issue of different speed between X and Y with Marlin 2.0.9.2 firmware (from download marlin web site). the Firmware Marlin-bugfix-2.0.9.2.x from https://github.com/bigtreetech/SKR-2/tree/master/Firmware has resolved this issue.

But now, with this bugfix firmware, I also have this issue with homing X or Y
I try to use sensorless endstops (SPI_ENDSTOPS)

When I try to Home X, Steppers A&B move to the Left (Sensorless Endstops X works fine) but steppers continue to move to the rear and don’t stop
If I try to Home Y, Steppers A&B move to the front (Endstops Y also works) but steppers continue to move to the right and don’t stop

PS:  I had to switch A&B steppers connectors to have correct direction on axis X & Y, because at least X axis was inverted, and this can’t be reverse by using invert settings in CoreXY or CoreYX mode.

closed 2022-03-27 by github-actions[bot]