MarlinFirmware/Marlin #23089: [BUG] Cant compile with DGUS_LCD_UI_RELOADED set as LCD

harrisck, 2021-11-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When selecting DGUS_LCD_UI_RELOADED as an LCD option in the Config the build fails with the following error

Compiling .pio/build/STM32F103RET6_creality/src/src/lcd/extui/dgus_reloaded/definition/DGUS_VPList.cpp.o
In file included from Marlin/src/lcd/extui/dgus_reloaded/../../../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin/src/lcd/extui/dgus_reloaded/../../../inc/../HAL/HAL.h:30,
                 from Marlin/src/lcd/extui/dgus_reloaded/../../../inc/MarlinConfig.h:31,
                 from Marlin/src/lcd/extui/dgus_reloaded/../ui_api.h:45,
                 from Marlin/src/lcd/extui/dgus_reloaded/DGUSScreenHandler.h:29,
                 from Marlin/src/lcd/extui/dgus_reloaded/DGUSScreenHandler.cpp:27:
Marlin/src/lcd/extui/dgus_reloaded/DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::SetStatusMessage(FSTR_P, millis_t)':
Marlin/src/lcd/extui/dgus_reloaded/DGUSScreenHandler.cpp:412:73: error: 'msg' was not declared in this scope; did you mean 'fmsg'?
  412 |   dgus_display.WriteStringPGM((uint16_t)DGUS_Addr::MESSAGE_Status, FTOP(msg), DGUS_STATUS_LEN, false, true);
      |                                                                         ^~~
Marlin/src/lcd/extui/dgus_reloaded/../../../inc/../HAL/./STM32/../shared/Marduino.h:96:48: note: in definition of macro 'FTOP'
   96 | #define FTOP(S) (reinterpret_cast<const char*>(S))
      |                                                ^
*** [.pio/build/STM32F103RET6_creality/src/src/lcd/extui/dgus_reloaded/DGUSScreenHandler.cpp.o] Error 1

### Expected behavior

The build to complete and be used after flashing the correct DWIN_SET to the screen

### Actual behavior

Build failed

### Steps to Reproduce

Tested in 2.0.9.2 and in bugfix branch

Just remove existing LCD config and select

#define DGUS_LCD_UI_RELOADED

### Version of Marlin Firmware

2.0.9.2

### Printer model

Ender 6

### Electronics

Stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

ellensp, 2021-11-07:
Looks to just be a typo in Marlin/src/lcd/extui/dgus_reloaded/DGUSScreenHandler.cpp
`  dgus_display.WriteStringPGM((uint16_t)DGUS_Addr::MESSAGE_Status, FTOP(msg), DGUS_STATUS_LEN, false, true);
`  
Should be 
`dgus_display.WriteStringPGM((uint16_t)DGUS_Addr::MESSAGE_Status, FTOP(fmsg), DGUS_STATUS_LEN, false, true);`

ellensp, 2021-11-07:
The fix is now in bugfix version of marlin

closed 2021-11-07 by ellensp