MarlinFirmware/Marlin #23091: [BUG] (bug summary)

gnugu13, 2021-11-07:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

by using toolmenu --> icon "Autolevel" the printer got the 3 x 3 plane. but no submenu on the screen
for the setting the z-nozzle to 0 (paper method).
i made the configuration like on that (wiki)site:
https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/wiki/3D_TOUCH

### Bug Timeline

11/07/2021

### Expected behavior

expect to get a new scrren after touching autolevel

### Actual behavior

no new screen after touch autolevel

### Steps to Reproduce

1. toolmenu -> Autolevel
2. the print measured the plane 3 x 3
3. waiting for the sub sccren for setting the z-nozzle to 0 (paper method)

### Version of Marlin Firmware

CONFIGURATION_H_VERSION 02000902

### Printer model

woodpecker EZT3D (??)

### Electronics

stock Robin Nano V1.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

platformio.ini
```
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = mks_robin_nano35
include_dir  = Marlin

``
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7492088/Marlin.zip)
`

ellensp, 2021-11-07:
You leave out the most important information. What type of LCD do you have ?

thisiskeithb, 2021-11-07:
> You leave out the most important information. What type of LCD do you have ?

It's hidden in the .zip link caught up in some markdown: `MKS_ROBIN_TFT35`.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7492088/Marlin.zip)

I would guess that menus have changed a bit since that tutorial was made, but it's been a while since I've ran the latest MKS UI.

closed 2022-01-18 by github-actions[bot]