MarlinFirmware/Marlin #23092: [BUG] Invenzione di gcode non esistenti

Emaxenvy, 2021-11-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Mando in stampa un gcode e dopo poco, totalmente a caso comprare un errore che mostra chiaramente questo=
Unknou command: "F1 X49.059 Y103.398 E0.02036"

Ho provato a cambiare sd e adattatore sd ma non funziona, ho anche formattato tutto più volte in fat32 ma niente da fare. Ho provato ad aggiornare la scheda ma con il marlin bugfix-2.0.x ma niente. Secondo me la scheda o non so cosa legge male il gcode che dovrebbe leggere questo 

"G1 X49.059 Y103.398 E0.02036"
Infatti se cerco il gcoce nella scheda ovvero questo "F1 X49.059 Y103.398 E0.02036" non me lho trova

### Bug Timeline

Nuovo bug

### Expected behavior

[Configuration.h Configuration_adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/7493026/Configuration.h.Configuration_adv.h.zip)
[image]

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Printer model

Prusa fai da te

### Electronics

Skr 1.4 

### Add-ons

tft35_v2.0.26 aggiornamento ultimo 11 aprile 2020

### Bed Leveling

ABL Linear grid

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

ellensp, 2021-11-08:
Provided config are for version #define CONFIGURATION_H_VERSION 020005
we are currently at  CONFIGURATION_H_VERSION 02000902
Please provide the bugfix versions of config files you used.

GMagician, 2021-11-08:
It seems a wrong read on sd. F and G have just 1 bit different, bit 0. Maybe some "on limit" timings?

Sembra una lettura errata sull'SD. F e G hanno solo un bit di differenza, il bit 0. Forse una temporizzazione al mlimite?

Emaxenvy, 2021-11-08:
E' successo di nuovo con il firmware aggiornato lascio di nuovo il file. Secondo voi può essere un problema della scheda sd?

closed 2022-01-18 by github-actions[bot]