MarlinFirmware/Marlin #23094: [BUG] BTT SKR E3 RRF. part cooling fan not working. works in 2.0.9.1 with same settings

BobCraig10, 2021-11-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Tried to turn on part cooling fan through display and gcode. no part cooling fan. 

### Bug Timeline

Only with 2.0.9.2 

### Expected behavior

I expect the part cooling fan to turn on 

### Actual behavior

part cooling fan does not come on

### Steps to Reproduce

Turn on fan with display
turn on fan through terminal

### Version of Marlin Firmware

2.0.9.2 bugfix

### Printer model

ender 3

### Electronics

BTT SKR E3 RRF

### Add-ons

BTT TFT70

### Bed Leveling

ABL Bilinear mesh

### Host Software

OctoPrint

ellensp, 2021-11-08:
Are you using configs from https://github.com/MarlinFirmware/Configurations/tree/release-2.0.9.2/config/examples/Creality/Ender-3/BigTreeTech%20E3%20RRF ?

BobCraig10, 2021-11-08:
> Are you using configs from https://github.com/MarlinFirmware/Configurations/tree/release-2.0.9.2/config/examples/Creality/Ender-3/BigTreeTech%20E3%20RRF ?

Yes, and checked the pin # in the pins file, it is correct.

hounslg, 2021-11-09:
I've noticed the same behaviour with skr mini e3 v2 when built using the non-maple build options e.g. STM32F103RE_btt_USB.  If I use the _maple options the fan works as expected.

Not had a chance to dig into it or log an issue, need to validate my config first to make sure I'd not done something daft.  I'm using the bugfix config files.

arminth, 2021-11-09:
I realized a similar issue when compiling latest bugfix for a BTT SKR 1.1 Pro board. Part-cooling fan and also controller-fan are not working anymore. Hotend-Cooloing ins auto-mode still  works. 
Had a working version dated September 30. Flashed that one and both fans work again. So no hardware issue.
 
I tried it with a clean bugfix branch and reworked the original configuration file for anycubic Kossel Linear plus but still the same effect.

Here are my configuration and pin-files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7504393/Marlin.zip)

arminth, 2021-11-10:
I want to add that I found a similar thing on my Ender 3 with an SKR 1.4 Turbo board! Here, the partfan was working but when trying to print, a got a big clogg finding that the hotend cooling fan was not running! I reverted to a september bugfix-2.0.x frmware and the effect was gone. Printer worked again after declogging (RANT!!).
So there seems to be a bigger issue with the fancontrolling in general.
I add the configuration for the ender/SKR1.4 as well for debugging.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7511850/Marlin.zip)

descipher, 2021-11-11:
Bug confirmed on any STM32Fx,  LCP176x and  AVR 
set_pwm_duty  is not working at the moment we are looking at it.

tpruvot, 2021-11-11:
its working here for the fan... but maybe only for that :p

BobCraig10, 2021-11-11:
> Bug confirmed on any STM32Fx, LCP176x and AVR set_pwm_duty is not working at the moment we are looking at it.

I have not had a chance to look farther into it, but around line 599 in configuration_adv.h there is a #define Cooler_fan_pin line that is not in the previous versions. does the pin number need to be added here even though it is in the pins file maybe? or is this maybe the cause?   Just trying to help
[image]
.

descipher, 2021-11-11:
HAL_STM32 uses a different call. STM32F103 uses __STM32F1__

tpruvot, 2021-11-11:
no, F1 can use both, its why i said with STM32 HAL...

descipher, 2021-11-11:
I wrote that code, Cooler Fan is for a chiller, like on a laser cutter etc.

descipher, 2021-11-11:
> no, F1 can use both, its why i said with STM32 HAL...
That's helpful to know is could use either.

tpruvot, 2021-11-11:
actually, now i use STM32 HAL for that, doesn't work on F1 HAL with the incomplete F103VE maple variant

descipher, 2021-11-11:
```
PinName pin_name = digitalPinToPinName(pin);
  TIM_TypeDef *Instance = (TIM_TypeDef *)pinmap_peripheral(pin_name, PinMap_PWM);

  uint16_t adj_val = Instance->ARR * v / v_size; <- this is duty update without changing the frequency
  if (invert) adj_val = Instance->ARR - adj_val;
```
This is why it works on HAL_STM32

tpruvot, 2021-11-11:
was only possible with the help of buildroot/share/PlatformIO/variants/MARLIN_F103VE_LONGER/PeripheralPins.c
and variant.h
// Timer Definitions
// Leave TIMER 2 for optional Fan PWM
#define TEMP_TIMER              3
// Leave TIMER 4 for TFT backlight PWM or Servo freq...
#define STEP_TIMER              5
#define TIMER_TONE              TIM6
#define TIMER_SERVO             TIM7
#define TIMER_SERIAL            TIM8

tpruvot, 2021-11-11:
im not sure how to fill the missing timer array for the F1 maple "board/variant"

descipher, 2021-11-11:
Do you mean what pin to timers are mapped for that board/chip?

tpruvot, 2021-11-11:
those are needed : https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/variants/MARLIN_F103VE_LONGER/PeripheralPins.c#L157-L159

descipher, 2021-11-11:
I see why you are not sure .. lol

arminth, 2021-11-11:
@thisiskeithb 
Maybe, you can add the "bug confirmed button" and change the Headline to a more general "Fans don't work on STM32Fx, LCP176x and AVR.  set_pwm_duty is not working."

@descipher 
Thanks for confirming! I started loosing my last hairs hinting that one down!

Thanks!

ManuelMcLure, 2021-11-12:
I think I just ran into this on Re-ARM/RAMPS. However, it seems to work on 2.0.9.2, but _not_ on current bugfix-2.0.x.

descipher, 2021-11-12:
> I think I just ran into this on Re-ARM/RAMPS. However, it seems to work on 2.0.9.2, but _not_ on current bugfix-2.0.x.

Yes, I hit it on the Re-Arm(LCP1768) #23110 addressed it. Looks like a revert is in queue ..

descipher, 2021-11-12:
At this point the STM32 still needs to be solved, it is more complicated than the NXP and AVR. I think it can be solved reasonable soon. I am 40% there. Or it can be reverted but the frequency challenge eventually needs to be addressed.

roffe, 2021-11-12:
I had same problems on MKS Robin nano 1.2 board, it's #23048 that broke part cooling, if i build bugfix-2.0.x from commit `767a15d468ab6859eec19600541ad21dff757d1a` (before 23048 was merged it works again 8

Jerronimo2001, 2021-11-12:
I have the same problem with BTT Octopus 1.0. "AUTO_FAN_PIN" or "CONTROLLER_FAN" works on every defined PIN but FAN selection from MENU or with terminal - no luck.

Kimmy2000, 2021-11-12:
Just want to say I have the same issue with the SKR Mini E3 1.2, part cooling fan is not working.

GJSchaller, 2021-11-13:
Confirming same behavior on my SKR mini E3 v2.0.  I remembered this from #20670 so I went back and disabled my "#define speaker", and made sure I wasn't using Maple... but still no fan.  I re-flashed a known good copy of 2.0.7.2 and confirmed my hardware works.

sarvenn, 2021-11-14:
I have an Ender 3 Pro / SKR Mini E3 V2 / BTT TFT35 E3 V3.
I compiled (with AutoBuildMarlin) latest bugfix 2.
Part cooling fan is not working.
Board fan is not working.
I checked my fans, no hardware problem.
So latest PRs did not solve the problem. I will go back to a previous build.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7534218/Marlin.zip)
.

sarvenn, 2021-11-15:
I uploaded Marlin 2.0.9.2 fw. Both fans are working.

BobCraig10, 2021-11-19:
Is this fixed or still in the process?

descipher, 2021-11-19:
> Is this fixed or still in the process?

#23125 was updated if you would like to test it.

silverdragonhawk, 2021-11-19:
@descipher how do I go about testing it? just copy and paste the new file to the old one? never done this before and don't want to mess it up.

DavisPoGo, 2021-11-19:
For those following this issue, the PR from @descipher fixes the part cooling fan function. I'm running an Ender 3 v2 with SKR Mini E3 v2

JohnnyTheOne, 2021-11-21:
I have the same problem with BTT GTR

thisiskeithb, 2021-11-21:
@thinkyhead Can you rename this to something like “Fans don’t work in STM32 HAL for some boards”?

npmomchev, 2021-11-22:
I have the same problem with BTT SKR PRO v1.1

thisiskeithb, 2021-11-22:
To anyone else finding this thread, the fan / pwm bug is already confirmed.

Apply the patch from https://github.com/MarlinFirmware/Marlin/pull/23125 or wait for it to be merged into the latest `bugfix-2.0.x`.

Jerronimo2001, 2021-11-23:
Exist sollution for "stm32f4" - SKR Octopuss ???

descipher, 2021-11-23:
> 

I don't see any reason why #23125 would not work for the STM32F446_5VX.

thisiskeithb, 2021-11-23:
#23125 has been merged.

closed 2021-11-23 by thisiskeithb